repetier/RepetierFirmware #601: Push-Pull Steppers

wfpaez, 2016-11-24:
I am attempting to configure my board (Arduino Due/RADDS) so that there are three (or four) steppers feeding filament - this is what the Diamond configuration.h does.  However, I want to use a conventional, non-Diamond hotend and I have no intention of mixing  In my configuration, the three filament feeders would feed in to a merging assembly and only one filament at a time would then be fed in to a fourth stepper.  This fourth stepper (presumably, E0) would be in the same configuration as a MK8/MK10 hotend.  Basically, I'm taking a printer with a single nozzle and, when I want to use a different filament, having the active extruder retract and become inactive, and have another extruder become active and feed itself in to E0.

I've seen 'Mixing Extruder' option, as well as, 'Extruders share same heater from E0.'  Mixing Extruder is close to what I want, but does that mean that E0 is the 'nozzle/heater' and that any stepper attached to it will mirror E1, E2, E3 if any of them are running (without enabling motor mirroring)?

I currently have E0 active and have added E1.  When I print (using E0 as the tool), E1 pulls filament and feeds it to E0.  E0 then pushes it to the hotend.  This is the behavior I want - two steppers pushing a piece of filament.  When I added E2 and sent a print (again, E0 is the active tool), E1 doesn't feed.  E2 does nothing as well.  I also tried making the tool be E1, but nothing happened there either.  I'm assuming I have to include a script at this point to indicate E1 or E2 should be pulling filament. 

Please let me know if none of this is making sense.  My goal is to be able to have multiple filaments (similar to the bowden tube machine), but with the flexibility of being able to use the more brittle blended PLA filaments, making E0 at the hotend necessary.

vinmicgra, 2016-11-24:
I've done this a couple weeks ago. You can do this by colormixing:

Color1: 50% E1 and 50% E0
Color2: 50% E2 and 50% E0
Color3: 50% E3 and 50% E0
Color4: 50% E4 and 50% E0

Now you can choose Color[1-4] and extrude with the according steppers. As a result of this misuse you have to double your steps/mm.

vincent.

wfpaez, 2016-11-24:
That's in Repetier-Host, yes?  I figured it was going to be the same as the Diamond, but with 0% of the unwanted feeder (which is what you are showing).

vinmicgra, 2016-11-24:
I've had to wipe my eprom after enabling the color mixing feature.

M502
M500

wfpaez, 2016-11-24:
Okay.  I'm brewing a fresh pot of really cheap, bad coffee and I will give it a try.  Thank you!