repetier/RepetierFirmware #602: Z_PROBE_Z_OFFSET_MODE = 1 bed coating confusion

luke321, 2016-11-25:
Hello,

I use a BL-Touch to measure bed level, I currently have one glas without coating and one with a 1mm thick print bite.

I also use bed coating thickness to level the bed because of different nozzle lengths.
So currently I have this odd behaviour:
Glas with coating: Bed coating thickness = 2mm Probe measures 10 mm with M30
Glas without coating: Bed coating thickness = 1mm Probe measures 8 mm with M30 !!!

The glasses both have the same thickness the coating is 1mm thick.

So I know this comes from this part of code, but I don't get it.
If I home my printer and move it to Z0, the bed coating value is already take into account.
This way I corrected my Z_offset before I even had a probe, so why is it added again?

#if Z_PROBE_Z_OFFSET_MODE == 1
distance += EEPROM::zProbeBedDistance();
#endif

repetier, 2016-11-26:
Ok, here description how it is supposed to work:

There are 2 ways to consider a changing bed coating, which are defined by Z_PROBE_Z_OFFSET_MODE.
Z_PROBE_Z_OFFSET_MODE = 0 means we measure the surface of the bed below any coating. This is e.g. 
the case with inductive sensors where we put BuildTak on top. In that case we can set Z_PROBE_Z_OFFSET
to the thickness of BuildTak to compensate. If we later change the coating, we only change Z_PROBE_Z_OFFSET
to new coating thickness.

Z_PROBE_Z_OFFSET_MODE = 1 means we measure the surface of the coating, e.g. because we have a mechanical switch.
In that case we add Z_PROBE_Z_OFFSET for the measured height to compensate for correct distance to bed surface.

In homing to max we reduce z length by Z_PROBE_Z_OFFSET to get a correct height.
In homing to z min we assume z endstop is bed level so we move up Z_PROBE_Z_OFFSET after endstop is hit. This 
requires the extruder to bend the coating thickness without harm!

So what is happening. Mode 1. We measure the bed. So lets say we measured 5.6mm and bed coating is 1mm so bed distance is 6.6mm so adding 1mm here is ok as the bed is 1mm away from measuring point. So here it follows what is expected probing should do.

So the question is more how do you switch between the coatings meaning what commands do you use and do they what is expected. Looking for this I saw at least that distortion correction will most probably mess up things adding the coating distance twice. So that needs to be fixed. Lets assume we are not using distortion correction for the moment.

To do your comparison you need to go to a defined Z and then run G30 and should get a difference from 1mm which is difference in bed coating, agreed. There are 2 ways to get a z position set - homing and G32 and both might mix up with the bed coating. So if you had home first and then G32 you might get a extra bed coating if there is an error. So lets see what I have.

Homing: This differs from drive system, min or max homing, bed coating and if z min = z probe. For this I assume cartesian with z min = z probe. Here I see an error that is always adds bed coating. So you are getting too high after homing as this should only be done with method 0.

G32: First we measure a regression plane using G30 distances, so including bed coating. The correct autolevel computes rotation, which is independent of constant z and also computes z which now also includes bed coating. At least for mode 1 it should subtract it.

So in total I found some errors for your combination that might even explain the 2mm difference. Hope to update soon, but need some more tests as I'm in a bigger change currently.

luke321, 2016-11-28:
Thank you for your detailed explanation, so basically there are some errors in the code.
My problem is created by using my Z-Min Switch to home and then using the Probe to measure the distance.

I removed the addition in this line of code for now, and everything works for me.
`#if Z_PROBE_Z_OFFSET_MODE == 1
distance += EEPROM::zProbeBedDistance();
#endif`

Thank you!

closed 2016-11-28 by luke321