repetier/RepetierFirmware #606: TEMP_2_PIN with 4 or more extruders

fobetor, 2016-12-02:
Hi,
 I have been testing the problem for a whole day, so I am sure there is something that I missed or that is wrong in the frimware.
The issue is about TEMP_2_PIN (Analog 15):
a) With 1,2,3 extruders everything works fine. The Thermistor gives the correct temp etc... for all T0,T1,T2
b) If I try to configure 4 or 5 extruders (I have X3 & Extrudrboard) then the TEMP_2_PIN with ANY analog pin number does not work (includin A15). Any thermistor plugged into it gives def.
If I plug the termistors to any other pin T0, T1 or others I defined for the extrudrboard (In AUX2) they work fine.
 If I change the pin assignment for the extruder in T2, to use T1 (the heat bed one) it works fine.
So.. anything I plug in T2 with more than 3 extruders does not work. It seems that when I define I have 4, 5,... extruders, T2 is dissabled...
I am getting mad along the code to try to figure what is going on .. but .. no luck
Any help please?
Thanks

repetier, 2016-12-02:
It is not possible to disable single thermistors. In fact we do not even use the symbolic names just the analog pin numbers. SO assigning extruder 15 ot TEMP_2_PIn does not make any difference. 
Having a extruder with thermistor should add it to ANALOG_INPUTS measured in HAL.cpp
In extruder.cpp you have
`const uint8 osAnalogInputChannels[] PROGMEM = ANALOG_INPUT_CHANNELS;`
where the pin numbers get stored. So they must be in ANALOG_INPUT_CHANNELS compite din Repetier.h

```
#define ANALOG_INPUTS (EXT0_ANALOG_INPUTS+EXT1_ANALOG_INPUTS+EXT2_ANALOG_INPUTS+EXT3_ANALOG_INPUTS+EXT4_ANALOG_INPUTS+EXT5_ANALOG_INPUTS+BED_ANALOG_INPUTS+THERMO_ANALOG_INPUTS+KEYPAD_ANALOG_INPUTS)
#if ANALOG_INPUTS > 0
/** Channels are the MUX-part of ADMUX register */
#define  ANALOG_INPUT_CHANNELS {EXT0_ANALOG_CHANNEL EXT1_ANALOG_CHANNEL EXT2_ANALOG_CHANNEL EXT3_ANALOG_CHANNEL EXT4_ANALOG_CHANNEL EXT5_ANALOG_CHANNEL BED_ANALOG_CHANNEL THERMO_ANALOG_CHANNEL KEYPAD_ANALOG_CHANNEL}
#endif
```

further atop these get computed to be empty/0 or channel and 1. Follow this up to your configuration variables and see how it comes.

For debugging use M105 X0 to get also raw values.

What happens if you assign the extruder having TEMP_2_PIN a other working thermistor. Is it still defect? It would follow the same chain and just end up with different analog channel.

Maybe the thermistor type for that extruder is simply wrong? Could not completely follow your swapping explainations.

fobetor, 2016-12-02:
Hi
Thank you for answering.
 I get a bit lost with the above explainations (I am not a programmer).
The extruder(s) are correct and work fine.
The figure is:
Define that I have 3 extruders (2 in RAMPS 1.4 and one in extrudrboard). Assign the default TEMP_0_PIN , TEMP_1_PIN,TEMP_2_PIN to extruder 1, bed & extruder 2 (Defaults from repetier) AND my own pins for the third extruder from AUX2 (They work fine)
With this config, I have all the thermistors working fine.
Now... If I say to repetier that I Have 4 or 5 extruders with the above PINS and assign new AUX2 Pins for extrdrboard extruder 4 , the result is that TEMP_2_PIN appears as def.
 If I swap the thermistor pin definition for extruder1 from TEMP_2_PIN to TEMP_1_PIN (bed) It works, but the Bed does not work.
 I can continue swaping these TEMP_2_PIN assigning it to other extruders or bed, but It will never work...
But the thermistors are correct and functional.
The other thermistors for the other extruders give me the correct temp. The problem arrises when I have to use TEMP_2_PIN.
 Even, If I assign ie: TEMP_2_PIN to Analog pin A3 and plug there the thermistor, I get no reading.
If I do the same with other thermistor pin ie: TEMP_3_PIN , I can read the temp.
So.. the error must be in the code...no?

fobetor, 2016-12-02:
I am using printrboard extrdrboard to have those extra extruders
No, you misunderstand me. It's easier.
With 3 extruders everything works fine
With 4 extruders, TEMP_2_PIN gets disabled for any thermistor.
That easy... even using a firmware configured throught your web page.

repetier, 2016-12-02:
That would mean the 4th extruder uses the pin as digital pin. I have a azteeg X3 Pro with 4 extruders and bed working so in general it works. Just not with your combination.

Question remains still how you get the 3rd/4th extruder configured as ramps does not have the drivers. I understand they are on the extension board but you need to define to define the pin mapping correctly which is where I think the problem lies. In config tool you might select the sockets but they are not defined in pins.h which you must have modified to achieve this.

fobetor, 2016-12-02:
The extruders uses the pins as analog.
How?
like this
http://reprap.org/wiki/Adding_more_extruders

BUT.. I maintain that more than 3 extruders in RAMPS 1.4 disables PIN_TEMP_2 ....

fobetor, 2016-12-02:
BTW, I can get the temps correctly from the thermistors in extrudrboard. using A11 & A12 from AUX2
The problem is with A15  pin

repetier, 2016-12-02:
A15 is digital pin 69

Try assigning extruder 4 and 5 E1 steppers and see if it changes anything.
Mapping for these extruders in RAMPS is mixed with azteeg pins so maybe you made a fault there.The test should show it.

fobetor, 2016-12-02:
Thanks. I will try and comment the results... :-)

fobetor, 2016-12-15:
At last I got some time to llok at this. The error was a type mysmatch in a pin numbering. Thanks to point me that 15 was 69 , that helped me to find the error (I didnt know that). 
thank you.

closed 2016-12-16 by repetier