repetier/RepetierFirmware #607: Some commands skipped

Kulitorum, 2016-12-05:
I use CuraEngine in Reptier host.
My printer has two heads, in a dual_x_axis configuration

I have put this in the "Before extruder switch"
G1 E-20 F3000 ; Build filament pressure
G92 E0
G1 E-20 F3000 ; Build filament pressure
G92 E0

And in "After extruder switch"
G92 E0
G1 E20 F3000 ; Build filament pressure
G92 E0

Problem is that with currect dev firmware, the commands are executed at wrong times, so what happens is:

Loop:
{
Extruder 0->1
Retraction to E-20 works.
heads switch place
Un-retraction E20 works
....printing....
Extruder 1->0
Retraction to E-20 works.
heads switch place
Un-retraction E20 NEVER HAPPENS
....printing... (without filament for the first E20)
)
goto loop :)

So on change from E1 to E0, the first unretract never happens but on change from E0 to E1, it works.

Ideas?

repetier, 2016-12-06:
My idea is that the T0 SELECT_COMMAND is not the same as for T1. It's the same routine just with variables filled with current extruder and since T1 is working the functions seem to be ok, only T0 with wrong definition.

Kulitorum, 2016-12-06:
So any suggestion what I could do, or are you taking a look at it?

repetier, 2016-12-06:
As I said I think it is in your configuration.h so you have to look if you have configured T0 correctly like T1, which I think you did not.

Kulitorum, 2016-12-06:
Looks correct to me - anything specific I should look for?

/*
This file is part of Repetier-Firmware.

Repetier-Firmware is free software: you can redistribute it and/or modify
it under the terms of the GNU General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

Repetier-Firmware is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU General Public License for more details.

You should have received a copy of the GNU General Public License
along with Repetier-Firmware.  If not, see <http://www.gnu.org/licenses/>.

*/

#ifndef CONFIGURATION_H
#define CONFIGURATION_H

/**************** READ FIRST ************************

This configuration file was created with the configuration tool. For that
reason, it does not contain the same informations as the original Configuration.h file.
It misses the comments and unused parts. Open this file file in the config tool
to see and change the data. You can also upload it to newer/older versions. The system
will silently add new options, so compilation continues to work.

This file is optimized for version 0.92
generator: http://www.repetier.com/firmware/v092/

If you are in doubt which named functions use which pins on your board, please check the
pins.h for the used name->pin assignments and your board documentation to verify it is
as you expect.

*/

#define NUM_EXTRUDER 2
#define MOTHERBOARD 80
#include "pins.h"

// ################## EDIT THESE SETTINGS MANUALLY ################
// ################ END MANUAL SETTINGS ##########################

#undef FAN_PIN
#define FAN_PIN ORIG_FAN2_PIN
#undef FAN_BOARD_PIN
#define FAN_BOARD_PIN -1
#define BOARD_FAN_SPEED 255
#define BOARD_FAN_MIN_SPEED 0
#define FAN_THERMO_PIN ORIG_FAN_PIN
#define FAN_THERMO_MIN_PWM 0
#define FAN_THERMO_MAX_PWM 255
#define FAN_THERMO_MIN_TEMP 45
#define FAN_THERMO_MAX_TEMP 60
#define FAN_THERMO_THERMISTOR_PIN TEMP_0_PIN
#define FAN_THERMO_THERMISTOR_TYPE 1

//#define EXTERNALSERIAL  use Arduino serial library instead of build in. Requires more ram, has only 63 byte input buffer.
// Uncomment the following line if you are using Arduino compatible firmware made for Arduino version earlier then 1.0
// If it is incompatible you will get compiler errors about write functions not being compatible!
//#define COMPAT_PRE1
#define BLUETOOTH_SERIAL  -1
#define BLUETOOTH_BAUD  115200
#define MIXING_EXTRUDER 0

#define DRIVE_SYSTEM 0
[... 1196 lines cut ...]
"moveZWhenHomed": "0",
"preheatTime": 30000,
"multiZEndstopHoming": "0",
"z2MinMaxPin": -1,
"z2MinMaxEndstop": 0,
"extruderIsZProbe": "0",
"boardFanMinSpeed": 0,
"doorPin": -1,
"doorEndstop": 1,
"uiAnimation": "0",
"hasMAX6675": false,
"hasMAX31855": false,
"hasGeneric1": false,
"hasGeneric2": false,
"hasGeneric3": false,
"hasUser0": false,
"hasUser1": false,
"hasUser2": false,
"numExtruder": 2,
"version": 100,
"primaryPortName": ""
}
========== End configuration string ==========

*/

qx1147, 2016-12-06:
Maybe you could post relevant snippets of the generated G-code, so we can see also the T-commands?
Moreover, you could try without comments at the end of command lines. I vaguely remember that I had problems with such kind of inline commenting - file was processed but with hiccups, and just moving those comments to separate lines fixed it.

Kulitorum, 2016-12-07:
This is the smallest Gcode I could come up with.

In line 76 when changing to T0, the filament is not un-retracted.

;Generated with Cura_SteamEngine 15.01
; Default start code
M140 S60 ;start heating bed
G90 ; Absolute positioning
M82 ; Extruder in absolute mode
G21        		;metric values
G28 ; Home extruder
M107 ; Turn off fan
M190 S60 ; wait for bed
; Activate all used extruder
M104 T0 S190
M104 T1 S190
G92 E0 ; Reset extruder position
; Wait for all used extruders to reach temperature
M109 T0 S190
M109 T1 S190
;G1 X5 Y5 Z1 F4000
;G1 E20 F400
;G1 E40 Z10 F400
;G92 E0
;Layer count: 4
;LAYER:0
M107
G0 F12000 X107.501 Y143.501 Z0.200
;TYPE:SKIRT
G1 F1440 X119.499 Y143.501 E0.66443
G1 X119.499 Y153.499 E1.21810
G1 X107.501 Y153.499 E1.88252
G1 X107.501 Y143.501 E2.43619
G0 F12000 X108.167 Y144.167
G1 F1440 X118.833 Y144.167 E3.02686
G1 X118.833 Y152.833 E3.50676
G1 X108.167 Y152.833 E4.09743
G1 X108.167 Y144.167 E4.57733
G1 F1200 E3.57733
G1 Z0.800
G0 F12000 X113.833 Y147.833
;TYPE:WALL-OUTER
G1 Z0.200
G1 F1200 E4.57733
G1 F840 X115.167 Y147.833 E4.65121
G1 X115.167 Y149.167 E4.72508
G1 X113.833 Y149.167 E4.79896
G1 X113.833 Y147.833 E4.87283
G92 E0
G1 F1200 E0.00000
G1 Z0.800
G92 E0
G1 E-20 F3000 ; Release filament pressure
G92 E0
T1
G92 E0
G1 E20 F3000 ; Build filament pressure
G92 E0
G0 F12000 X113.167 Y147.833
G1 Z0.200
G1 F1200 E0.00000
G1 F840 X113.167 Y149.167 E0.07387
G1 X111.833 Y149.167 E0.14775
G1 X111.833 Y147.833 E0.22162
G1 X113.167 Y147.833 E0.29550
;LAYER:1
M106 S127
G0 F12000 X113.167 Y147.833 Z0.500
;TYPE:WALL-OUTER
G1 F540 X113.167 Y149.167 E0.40631
G1 X111.833 Y149.167 E0.51712
G1 X111.833 Y147.833 E0.62793
G1 X113.167 Y147.833 E0.73874
G92 E0
G1 F1200 E0.00000
G1 Z1.100
G92 E0
G1 E-20 F3000 ; Release filament pressure
G92 E0
T0
G92 E0
G1 E20 F3000 ; Build filament pressure
G92 E0
G0 F12000 X113.833 Y147.833
G1 Z0.500
G1 F1200 E0.00000
G1 F540 X115.167 Y147.833 E0.11081
G1 X115.167 Y149.167 E0.22162
G1 X113.833 Y149.167 E0.33243
G1 X113.833 Y147.833 E0.44325
;LAYER:2
M106 S255
G0 F12000 X113.833 Y147.833 Z0.800
;TYPE:WALL-OUTER
G1 F540 X115.167 Y147.833 E0.55406
G1 X115.167 Y149.167 E0.66487
G1 X113.833 Y149.167 E0.77568
G1 X113.833 Y147.833 E0.88649
G92 E0
G1 F1200 E0.00000
G1 Z1.400
G92 E0
G1 E-20 F3000 ; Release filament pressure
G92 E0
T1
G92 E0
G1 E20 F3000 ; Build filament pressure
G92 E0
G0 F12000 X113.167 Y147.833
G1 Z0.800
G1 F1200 E0.00000
G1 F540 X113.167 Y149.167 E0.11081
G1 X111.833 Y149.167 E0.22162
G1 X111.833 Y147.833 E0.33243
G1 X113.167 Y147.833 E0.44325
;LAYER:3
G0 F12000 X113.167 Y147.833 Z1.100
;TYPE:WALL-OUTER
G1 F540 X113.167 Y149.167 E0.55406
G1 X111.833 Y149.167 E0.66487
G1 X111.833 Y147.833 E0.77568
G1 X113.167 Y147.833 E0.88649
G92 E0
G1 F1200 E0.00000
G1 Z1.700
G92 E0
G1 E-20 F3000 ; Release filament pressure
G92 E0
T0
G92 E0
G1 E20 F3000 ; Build filament pressure
G92 E0
G0 F12000 X113.833 Y147.833
G1 Z1.100
G1 F1200 E0.00000
G1 F540 X115.167 Y147.833 E0.11081
G1 X115.167 Y149.167 E0.22162
G1 X113.833 Y149.167 E0.33243
G1 X113.833 Y147.833 E0.44325
M107
G1 F1200 E-0.55675
G1 Z1.700
G0 F12000 X113.833 Y147.833 Z6.000
T0
; Default end code
;G1 Y0 ; Get extruder out of way. Uncomment to use!
G28 X0 Y0
M107 ; Turn off fan
; Disable all extruder
G91 ; Relative positioning
T0
G1 E-10 ; Reduce filament pressure
T1
G1 E-10 ; Reduce filament pressure
M104 T0 S0
M104 T1 S0
M140 S0 ; Disable heated bed
M84 ; Turn steppers off

Kulitorum, 2016-12-07:
Saving for SD print (and thereby stripping all the comments) makes no difference.

repetier, 2016-12-07:
Ok, I thought you had the command in firmware here
#define EXT0_SELECT_COMMANDS ""
#define EXT0_DESELECT_COMMANDS ""

looking at your gcode I see that both switches contain the correct code, so code generation is working as expected. I see that you have dual x axis, so maybe it is related to that fact and that you have dual x axis enabled also I do not see why this should happen. What happens if you run it manually and extrude after switch back, does that work? You said that later extrusion works, so from which command on does extruder suddenly work?

Kulitorum, 2016-12-08:
It's difficult to say when it starts working again because there's missing 20mm of filament, so all I know is that after about 20mm of feedstock, it runs again.

I had to rebuild the machine to do some large prints (removed the one head, reconfigured firmware) but I will complete these prints over the weekend and reconfigure to dual_X again. I will then try the manual experiments.