repetier/RepetierFirmware #611: Silent beep with full graphic controller

sanchosk, 2016-12-11:
Hi.
I am using firmware version 0.92, freshly downloaded from the website (not github, but using repetier configurator).
I bought a cheap chinese LCD 128x64 with encoder and buzzer.
The problem is - when using RAMPS with nothing else connected just the display, I get a short loud beep every 10-15 seconds.
When I disable beeper in configuration using option #define FEATURE_BEEPER 0, I can still hear it VERY SILENTLY beeping whenever I move the encoder.
It also makes the very silent sound once every few seconds. This can only be heard when laptop's fan is not working and everything else is silent, but still, it's SO annoying.
I am thinking about disabling it physically by either de-soldering it or by breaking the PCB trace, but still - it would be great if the software issue would be removed in the first place.

repetier, 2016-12-12:
The beeping comes from the sensor error since you have no thermistors connected.

The problem that it does not disable is that the controller it self defines the beeper overriding your setting. Maybe we should just make controller set pin but not enable beeper as function and leave that to configuration.h setting.

sanchosk, 2016-12-12:
So majority of the reason for the bug is between keyboard and chair, as usual :)
Will check the source, perhaps I can find the definition of the controller and disable the pin there.
However, it would be nice to have something like check if the beeper is disabled and then just not define the beeper pin...

sanchosk, 2016-12-12:
I made one small modification in ui.h - line 2110, I changed the condition to
#if BEEPER_TYPE==0 or FEATURE_BEEPER == 0
and now the beeper is disabled and no "silent noises" are produced :)
Thanks for the hint.

repetier, 2016-12-12:
or is not correct must be || so
#if BEEPER_TYPE==0 || FEATURE_BEEPER == 0

would be correct.

sanchosk, 2016-12-12:
Interesting - it compiled OK :)

repetier, 2016-12-12:
Just took a look at beep function which starts like that

```
void beep(uint8_t duration,uint8_t count)
{
#if FEATURE_BEEPER

```

so I'm wondering if that is new in dev version or why it beeped in your case as you had FEATURE_BEEPER 0 - so even when function gets called it should not make a sound. In any case for next release we have both as it reduces size.

closed 2016-12-12 by repetier