repetier/RepetierFirmware #615: The port is not responding after "Disable steppers after inactivity"

Sanitar777, 2016-12-17:
Hello!

I have the following configuration:
Arduino DUE + RADDS + RADDS LCD Display 4x20
H-bot/CoreXY
Repetier-Firmware DEV

If I set 'Disable steppers after inactivity after' to '360' and that time ends, as expected, the motors of the printer are switched off. But the port stops accepting any commands. Helps only restart the Arduino DUE.

If I set  'Disable steppers after inactivity after' to '0' - everything works fine, the port responds to commands. But the motors of the printer, never turn off.

Thank you!

repetier, 2016-12-17:
Any commands means M115 does also return nothing in log? No communication or do just motors not turn on?

All the timer dies is disabling steppers nothing communication related.

Sanitar777, 2016-12-17:
The port stops accepting **any** commands.

Sanitar777, 2016-12-17:
> **All the timer dies is disabling steppers nothing communication related.**
I understand that. I understand that it should be. But in fact says busy doing something and not responding to input any commands.

repetier, 2016-12-17:
You mean it always outputs 
busy: something

to host? That should prevent host from sending new commands and would be wrong. Timeout should not trigger busy mode. Will test it.

repetier, 2016-12-17:
OK tested it and for me nothing happens except that motors are off, so working as expected.

It is outputs busy I guess my changes have fixed it somehow (locally I'm a bit advance). If it is just doing nothing not reporting "busy:" the change might have caused a problem with power unit. Try if this also happens if the main power is off also you can not see motor off it still will happen.

Sanitar777, 2016-12-17:
[image]

15:10:36.237 : OpenGL version:4.0.0 - Build 10.18.10.4358
15:10:36.238 : OpenGL extensions:GL_EXT_blend_minmax GL_EXT_blend_subtract GL_EXT_blend_color GL_EXT_abgr GL_EXT_texture3D GL_EXT_clip_volume_hint GL_EXT_compiled_vertex_array GL_SGIS_texture_edge_clamp GL_SGIS_generate_mipmap GL_EXT_draw_range_elements GL_SGIS_texture_lod GL_EXT_rescale_normal GL_EXT_packed_pixels GL_EXT_texture_edge_clamp GL_EXT_separate_specular_color GL_ARB_multitexture GL_ARB_map_buffer_alignment GL_ARB_conservative_depth GL_EXT_texture_env_combine GL_EXT_bgra GL_EXT_blend_func_separate GL_EXT_secondary_color GL_EXT_fog_coord GL_EXT_texture_env_add GL_ARB_texture_cube_map GL_ARB_transpose_matrix GL_ARB_internalformat_query GL_ARB_internalformat_query2 GL_ARB_texture_env_add GL_IBM_texture_mirrored_repeat GL_EXT_multi_draw_arrays GL_SUN_multi_draw_arrays GL_NV_blend_square GL_ARB_texture_compression GL_3DFX_texture_compression_FXT1 GL_EXT_texture_filter_anisotropic GL_ARB_texture_border_clamp GL_ARB_point_parameters GL_ARB_texture_env_combine GL_ARB_texture_env_dot3 GL_ARB_texture_env_crossbar GL_EXT_texture_compression_s3tc GL_ARB_shadow GL_ARB_window_pos GL_EXT_shadow_funcs GL_EXT_stencil_wrap GL_ARB_vertex_program GL_EXT_texture_rectangle GL_ARB_fragment_program GL_EXT_stencil_two_side GL_ATI_separate_stencil GL_ARB_vertex_buffer_object GL_EXT_texture_lod_bias GL_ARB_occlusion_query GL_ARB_fragment_shader GL_ARB_shader_objects GL_ARB_shading_language_100 GL_ARB_texture_non_power_of_two GL_ARB_vertex_shader GL_NV_texgen_reflection GL_ARB_point_sprite GL_ARB_fragment_program_shadow GL_EXT_blend_equation_separate GL_ARB_depth_texture GL_ARB_texture_rectangle GL_ARB_draw_buffers GL_ARB_color_buffer_float GL_ARB_half_float_pixel GL_ARB_texture_float GL_ARB_pixel_buffer_object GL_EXT_framebuffer_object GL_ARB_draw_instanced GL_ARB_half_float_vertex GL_ARB_occlusion_query2 GL_EXT_draw_buffers2 GL_WIN_swap_hint GL_EXT_texture_sRGB GL_ARB_multisample GL_EXT_packed_float GL_EXT_texture_shared_exponent GL_ARB_texture_rg GL_ARB_texture_compression_rgtc GL_NV_conditional_render GL_ARB_texture_swizzle GL_EXT_texture_swizzle GL_ARB_texture_gather GL_ARB_sync GL_ARB_framebuffer_sRGB GL_EXT_packed_depth_stencil GL_ARB_depth_buffer_float GL_EXT_transform_feedback GL_ARB_transform_feedback2 GL_ARB_draw_indirect GL_EXT_framebuffer_blit GL_EXT_framebuffer_multisample GL_ARB_framebuffer_object GL_ARB_framebuffer_no_attachments GL_EXT_texture_array GL_EXT_texture_integer GL_ARB_map_buffer_range GL_ARB_texture_buffer_range GL_EXT_texture_buffer GL_EXT_texture_snorm GL_ARB_blend_func_extended GL_INTEL_performance_queries GL_INTEL_performance_query GL_ARB_copy_buffer GL_ARB_sampler_objects GL_NV_primitive_restart GL_ARB_seamless_cube_map GL_ARB_uniform_buffer_object GL_ARB_depth_clamp GL_ARB_vertex_array_bgra GL_ARB_shader_bit_encoding GL_ARB_draw_buffers_blend GL_ARB_geometry_shader4 GL_EXT_geometry_shader4 GL_ARB_texture_query_lod GL_ARB_explicit_attrib_location GL_ARB_draw_elements_base_vertex GL_ARB_instanced_arrays GL_ARB_base_instance GL_ARB_fragment_coord_conventions GL_EXT_gpu_program_parameters GL_ARB_texture_buffer_object_rgb32 GL_ARB_compatibility GL_ARB_texture_rgb10_a2ui GL_ARB_texture_multisample GL_ARB_vertex_type_2_10_10_10_rev GL_ARB_timer_query GL_ARB_tessellation_shader GL_ARB_vertex_array_object GL_ARB_provoking_vertex GL_ARB_sample_shading GL_ARB_texture_cube_map_array GL_EXT_gpu_shader4 GL_ARB_gpu_shader5 GL_ARB_gpu_shader_fp64 GL_ARB_shader_subroutine GL_ARB_transform_feedback3 GL_ARB_get_program_binary GL_ARB_separate_shader_objects GL_ARB_shader_precision GL_ARB_vertex_attrib_64bit GL_ARB_viewport_array GL_ARB_transform_feedback_instanced GL_ARB_compressed_texture_pixel_storage GL_ARB_shader_atomic_counters GL_ARB_shading_language_packing GL_ARB_shading_language_420pack GL_ARB_texture_storage GL_EXT_texture_storage GL_ARB_vertex_attrib_binding GL_ARB_multi_draw_indirect GL_ARB_program_interface_query GL_ARB_texture_storage_multisample GL_ARB_buffer_storage GL_ARB_debug_output GL_KHR_debug GL_ARB_arrays_of_arrays GL_INTEL_map_texture GL_ARB_texture_compression_bptc GL_ARB_ES2_compatibility GL_ARB_ES3_compatibility GL_ARB_robustness GL_EXT_texture_sRGB_decode GL_KHR_blend_equation_advanced GL_EXT_shader_integer_mix GL_ARB_stencil_texturing 
15:10:36.239 : OpenGL renderer:Intel(R) HD Graphics 4000
15:10:36.240 : Using fast VBOs for rendering is possible
15:11:27.229 : Printer reset detected - initalizing
15:11:27.246 : start
15:11:27.246 : Info:Software Reset
15:11:27.246 : Transformat00 0.000000 1.000000
15:11:27.246 : Free RAM:81744
15:11:27.261 : SelectExtruder:0
15:11:27.263 : FlowMultiply:100
15:11:27.263 : wait
15:11:27.263 : wait
15:11:27.263 : wait
15:11:27.263 : wait
15:11:27.263 : wait
15:11:27.263 : wait
15:11:27.263 : wait
15:11:27.263 : Printer reset detected - initalizing
15:11:27.263 : start
15:11:27.263 : Info:PowerUp
15:11:27.264 : Transformation matrix: 1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000
15:11:27.440 : N1 M110*34
15:11:27.440 : N2 M115*36
15:11:27.440 : N4 M114*35
15:11:27.450 : N5 M111 S6*98
15:11:27.452 : N6 T0*60
15:11:27.453 : N7 M80*28
15:11:28.490 : Free RAM:81744
15:11:28.490 : SelectExtruder:0
15:11:28.490 : FlowMultiply:100
15:11:28.490 : ok
15:11:28.494 : ok 2
15:11:28.509 : N9 M220 S100*104
15:11:28.509 : N10 M221 S100*81
15:11:28.509 : N11 M111 S6*87
15:11:28.509 : N12 T0*9
[... cut ...]
15:12:45.959 : busy:processing
15:12:47.957 : busy:processing
15:12:49.080 : X:-18.00 Y:0.00 Z:300.000 E:0.0000
15:12:49.080 : ok 42
15:12:49.086 : ok 43
15:12:49.092 : ok 44
15:12:49.098 : ok 45
15:12:49.104 : ok 46
15:12:49.110 : ok 47
15:12:49.116 : ok 48
15:12:49.121 : ok 49
15:12:49.127 : ok 50
15:12:49.133 : ok 51
15:12:49.139 : ok 52
15:12:49.144 : ok 53
15:12:49.150 : ok 54
15:12:49.157 : ok 55
15:12:49.162 : ok 56
15:12:50.112 : ok 57
15:12:51.111 : wait
15:12:52.115 : wait
15:12:53.114 : wait
15:12:53.172 : ok 58

Sanitar777, 2016-12-17:
What was he doing for almost a minute?

repetier, 2016-12-17:
The busy comes from your homing here. That is normal and during that period you can not send commands, that is correct.

Temperature graph looks wrong. Like temp. is dropping to 0 all the time and no heating at all.
Could not see what was with timeout as that log is not here. Also log seems to have swallowed the * where checksum starts.

Sanitar777, 2016-12-17:
I couldn't enter any commands! They are in port was not. Once worked off timer engines, began to ride temperature. The program then hangs.

Sanitar777, 2016-12-17:
See screenshot
'Communication timeout - reset send buffer block'

Sanitar777, 2016-12-17:
I gave the command to home. Everything is frozen. And in one minute, the command worked. How so? And you say that everything is fine.

Sanitar777, 2016-12-17:
If I set 'Disable steppers after inactivity after' to '0' - everything works fine, the port responds to commands. And temperature works correctly.

repetier, 2016-12-17:
You are mixing several issues here I think.

G28 -> busy = normal. Once homing is finished you can send commands again. That is no problem.

The timeout is a problem, but is not captured in the log you showed. 

All what a value != 0 does is this code:
        if(stepperInactiveTime != 0 && curtime >  stepperInactiveTime )
            Printer::kill(true);

and that just sets 3 enable pins for motors to false, nothing else. If that causes your communication to break it is maybe a hardware problem which is why a said to test without main power so no voltage drop from power unit can influence results. Some power units react bad on sudden current changes and that can have unwanted effects.

Also what happens if you turn motors off using M84 - does more or less the same.

Sanitar777, 2016-12-17:
I can certainly mix and that is you should know better. 
I'm sorry, I can't show that it is not considered in the log. 

I have the procedure home in 10 seconds maximum. Sorry. 

With the power I have everything in order too. More than enough. 
Additionally, the table and the extruder are not included. 
What is more, I tried different power sources, the result is the same.

Sanitar777, 2016-12-17:
No matter what I mixed what I'm doing wrong, I only know one:
If I set 'Disable steppers after inactivity after' to '0' - in this case everything starts to work correctly.

repetier, 2016-12-17:
Always hard to tell what is wrong if I only get part of required information:-)

What I see is G28 and 40s busy. Knowing homing causes busy I simply assume homing took 40s. And afterwards I see "ok" so communication goes on everything is looking fine.

Anyhow what does M84 do? Does it cause same problems?

All I can say is that my due does not have these problems when they timeout and no other user has reported the problem. And you did not make the tests I asked for (without power, M84 with power) so I can not say anything more.

Sanitar777, 2016-12-17:
> Anyhow what does M84 do? Does it cause same problems?
No problems.

repetier, 2016-12-17:
M84 does the same as timeout, means it calls the same function to disable motors. Since that works it is no power issue.

But it also means I have no idea why it does not work for you or why it causes a problem. So for now I suggest setting timeout to 0 as it works and add M84 as last command in end gcode so printer disables motors after end of print.

Sanitar777, 2016-12-17:
I have done so already. 
But this does not solve the problem. 
I am sure it will appear someone else. 

Thank you.

repetier, 2016-12-17:
Yes I know it is not that satisfying. But I currently do not see where it goes wrong for you or why it happens only to you and not to me. I know that often something that should not happen still does happen and often these problems go away with next update. These are the most difficult problems as the source is not where the problem appears. If you have any more insight or I see something we can proceed from there. But now I'm stuck.

Sanitar777, 2016-12-17:
It's okay, don't worry about this.
 
Also I think that with time everything will return to normal. And this problem will go away when all cleaned up in the code before the release of the firmware. 

I'll see you later, I'll find something else :-) 
Good Luck and thank you!

Sanitar777, 2016-12-17:
A temporary solution was found.

closed 2016-12-17 by Sanitar777