repetier/RepetierFirmware #617: Printing stops when using display

Kulitorum, 2016-12-17:
On the latest version (downloaded 2016-12-06) if the printer is printing over USB, and I use the Geetech full graphics display to turn on or off the display, printing stops when I am using the display. I have to "Force Next" on Simplify3d to continue printing.

repetier, 2016-12-18:
what does "to turn on or off the display" mean? The LCD display is always on, so I guess you mean some other function. So which function breaks communication?

Since force next helps I guess it just caused some communication error and Simplify3D does not use the hints from firmware to solve it on their own. But might of course also be a output with missing newline so ok was not in own line so knowing the exact function would help here as many functions return changed data to connected host.

Kulitorum, 2016-12-18:
Sorry, should have said "on or off the Fan".....

Maybe using the display causes an overload of the processor and it misses some USB data or something? - The problem is very consistent and happens every time I test it.

Today I am printing from SD, and I can use the display without issues.

Kulitorum, 2017-01-14:
Just touching the knob on the display will crash communication. I click tthe button, and printing continues, but as soon as I rotate the button, printing stops.

Kulitorum, 2017-01-14:
I just tried with latest dev version, the issue seems to have been resolved.

closed 2017-01-14 by Kulitorum

repetier, 2017-01-14:
Great that it is working again, also I'm not aware of any changes that makes communication fail. Just reworked menu structure, added some function.