repetier/RepetierFirmware #618: Z-axis randomly stops during print (filament on detection on)

sztojfen, 2016-12-23:
Hello,
Im using repetier firmware 0.92.9 with filament jam detection on. The printer itself works fine, but sometimes has some Z-axis issues. At random height it just suddenly stops. The other 2 axes work perfectly fine. After the print is cancelled, the z-axis is still NOT responsive.
-im using MKS 1.4
-the printer detects filament jam correctly
-the jam pin was originally set to z-max pin, I changed it to x-max pin, it didn't solve the problem
-i did not notice this problem before firmware upgrade (0.91 before)
-the ALWAYS_CHECK_ENDSTOPS is set to 0
-resetting the MKS solves the issue'(z-axis is "on the move" again)
-I did not manage to notice any patterns with this problem, it occurs on random height, random models, random slicers etc.

As resetting the MKS solves the problem, my guess is that its not a hardware issue.
Below is my config file:

#define NUM_EXTRUDER 1
#define MOTHERBOARD 33
#include "pins.h"

#undef FAN1_PIN
#define FAN1_PIN 9
#undef FAN2_PIN
    #define FAN2_PIN HEATER_1_PIN
//#undef FAN_BOARD_PIN
//#define FAN_BOARD_PIN 9 //HEATER_2_PIN
#define BOARD_FAN_SPEED 128
#define FAN_THERMO_PIN -1
#define FAN_THERMO_MIN_PWM 128
#define FAN_THERMO_MAX_PWM 255
#define FAN_THERMO_MIN_TEMP 45
#define FAN_THERMO_MAX_TEMP 60
#define FAN_THERMO_THERMISTOR_PIN -1
#define FAN_THERMO_THERMISTOR_TYPE 1

//#define EXTERNALSERIAL  use Arduino serial library instead of build in. Requires more ram, has only 63 byte input buffer.
// Uncomment the following line if you are using Arduino compatible firmware made for Arduino version earlier then 1.0
// If it is incompatible you will get compiler errors about write functions not being compatible!
//#define COMPAT_PRE1
#define BLUETOOTH_SERIAL  -1
#define BLUETOOTH_BAUD  115200
#define MIXING_EXTRUDER 0

#define DRIVE_SYSTEM 0
#define XAXIS_STEPS_PER_MM 160
#define YAXIS_STEPS_PER_MM 160
#define ZAXIS_STEPS_PER_MM 800
#define EXTRUDER_FAN_COOL_TEMP 50
#define PDM_FOR_EXTRUDER 0
#define PDM_FOR_COOLER 0
#define DECOUPLING_TEST_MAX_HOLD_VARIANCE 20
#define DECOUPLING_TEST_MIN_TEMP_RISE 1
#define KILL_IF_SENSOR_DEFECT 0
#define RETRACT_ON_PAUSE 6
#define PAUSE_START_COMMANDS ""
#define PAUSE_END_COMMANDS ""
#define SHARED_EXTRUDER_HEATER 0
#define EXT0_X_OFFSET 0
#define EXT0_Y_OFFSET 0
#define EXT0_Z_OFFSET 0
#define EXT0_STEPS_PER_MM 184
#define EXT0_TEMPSENSOR_TYPE 1
#define EXT0_TEMPSENSOR_PIN TEMP_0_PIN
#define EXT0_HEATER_PIN HEATER_0_PIN
#define EXT0_STEP_PIN ORIG_E0_STEP_PIN
#define EXT0_DIR_PIN ORIG_E0_DIR_PIN
#define EXT0_INVERSE 0
#define EXT0_ENABLE_PIN ORIG_E0_ENABLE_PIN
#define EXT0_ENABLE_ON 0
#define EXT0_MIRROR_STEPPER 0
#define EXT0_STEP2_PIN ORIG_E0_STEP_PIN
#define EXT0_DIR2_PIN ORIG_E0_DIR_PIN
#define EXT0_INVERSE2 0
[... 405 lines cut ...]
#define CASE_LIGHT_DEFAULT_ON 1
#define UI_START_SCREEN_DELAY 1000
#define UI_DYNAMIC_ENCODER_SPEED 1
        /**
Beeper sound definitions for short beeps during key actions
and longer beeps for important actions.
Parameter is delay in microseconds and the secons is the number of repetitions.
Values must be in range 1..255
*/
#define BEEPER_SHORT_SEQUENCE 2,2
#define BEEPER_LONG_SEQUENCE 8,8
#define UI_SET_PRESET_HEATED_BED_TEMP_PLA 60
#define UI_SET_PRESET_EXTRUDER_TEMP_PLA   190
#define UI_SET_PRESET_HEATED_BED_TEMP_ABS 110
#define UI_SET_PRESET_EXTRUDER_TEMP_ABS   240
#define UI_SET_MIN_HEATED_BED_TEMP  30
#define UI_SET_MAX_HEATED_BED_TEMP 120
#define UI_SET_MIN_EXTRUDER_TEMP   170
#define UI_SET_MAX_EXTRUDER_TEMP   260
#define UI_SET_EXTRUDER_FEEDRATE 2
#define UI_SET_EXTRUDER_RETRACT_DISTANCE 3

#define NUM_MOTOR_DRIVERS 0

#endif

chexov, 2016-12-23:
I can confirm having same issue.
Even during homing (XYZ) Z axis stops randomly. Even rewired all my signal and power cabels since I was thinking it was the noise. After a week of struggle I rollback to Marlin ad surprised it worked. Stayed there to have a working system.
Ready to test dev branch or PR.
:beer:

repetier, 2016-12-23:
How does this show up. Are it several moves the got skipped or is the z completely not working.
If not working does it make sounds like loosing steps? If you hear a buzz on moves the acceleration or z jerk is probably too high. Please also check what you have in eeprom compared to configuration.h as we use values in eeprom.

sztojfen, 2016-12-23:
The z axis is not working at all. There is no noise at all and no z movement at all. The x and y axes continue the print at the same height. 
It fells like the enable pin is off. Next week i will connect the stp and enable pins of z stepper driver to oscilloscope and post the results.
The firmware is freshly loaded, followed by m502 and m500 just to be sure, so the EEPROM values are same as in config.

Merry Xmas!

repetier, 2016-12-23:
Enable is easy to test. You can turn z axis easily by hand if enable is off. Would be a interesting test.

Alternatively it could be that it does not queue the move or abort directly if endstop is triggered but that would normally vanish for next move if it is a false signal.

Could some other function use the same pin (enable) and get triggered somehow when it happens?

In any case I never had that picture. Already curious what it turns out.

chexov, 2016-12-23:
I periodically update development branch just to help with testing on one of mine DIY UMO machines.
Last time I got this issue reproduced it was on 0e0d78b6267e527f3ed5f1857fe587d23233dc5d commit.
So issue appeared with no change in hardware for me.
Z limit switch is NC. Nothing else connected with it.

Sanitar777, 2016-12-25:
> is the z completely not working.
If not working does it make sounds like loosing steps?

I have the same problem on firmware Dev. I have the same problem on firmware Dev. During printing, the Z axis turns off and it make sounds like loosing steps.

Thank you.