repetier/RepetierFirmware #620: printer pauses with jam detection on filament change

ekaggrat, 2017-01-04:
this is a wierd error that i have been able to repeat 3 - 4 times consistently.. I have a filament jam sensor using a hall sensor which needs 2500 steps in normal mode and 3500 for jam detection.. The sensor works great and has saved me a few times.. The problem is that after i finish one print and for the next print i change the filament , then when the next print starts the printer prints around one line and then haults with a jam error.. The only way to get it going is to reset the arduino. I thought it was a problem with the jam sensor, but i have tried with two sensors and the same error occurs.. It is not a big deal just frustrating. 
The jam sensor is mounted near the spool about 450 mm from the direct drive extruder on my custom corexy.. 

thanks

repetier, 2017-01-04:
Since you manually change position the firmware does not see it and may count up to 5000 steps whcih would trigger jam detection. Please try sending
M513
after manually changing filament or in start script of prints. It clears all jam marker and sets count to 0. I think that will help.

ekaggrat, 2017-01-04:
ok great , i will do that.. anyway of clearing it from the lcd besides suffixing it in the GCODE file?

closed 2017-01-04 by ekaggrat

repetier, 2017-01-04:
I hope the filament change function would do that as well.