repetier/RepetierFirmware #621: Z_PROBE_X_OFFSET and Z_PROBE_Y_OFFSET Not available in EEPROM settings GUI

Kulitorum, 2017-01-06:
I cannot edit Z_PROBE_X_OFFSET in the user interface in repetier host, and I cannot change it in config.h

repetier, 2017-01-06:
Just tested it and could edit it in eeprom. With eeprom the config is ignored so that would be ok. What is the problem when editing the value in eeprom? Does it not appear or not save value? Reopening eeprom values for me showed the new value as expected.

Kulitorum, 2017-01-06:
I dont see it in the eepromconfig. Is this because I use a old version of host? - the latest takes 5 to connect to a printer because the firmware is spamming a lot on connect.

repetier, 2017-01-07:
No it should work with all hosts supporting eeprom. It always appears if FEATURE_Z_PROBE is 1

#if FEATURE_Z_PROBE
    writeFloat(EPR_Z_PROBE_HEIGHT, Com::tZProbeHeight);
    writeFloat(EPR_Z_PROBE_BED_DISTANCE, Com::tZProbeBedDitance);
    writeFloat(EPR_Z_PROBE_SPEED, Com::tZProbeSpeed);
    writeFloat(EPR_Z_PROBE_XY_SPEED, Com::tZProbeSpeedXY);
    writeFloat(EPR_Z_PROBE_X_OFFSET, Com::tZProbeOffsetX);
    writeFloat(EPR_Z_PROBE_Y_OFFSET, Com::tZProbeOffsetY);

what does "the latest takes 5 to connect to a printer because the firmware is spamming a lot on connect." mean?

Kulitorum, 2017-01-14:
Sorry, this was me being a idiot...

closed 2017-01-14 by repetier