repetier/RepetierFirmware #623: Z_PROBE_X_OFFSET and Z_PROBE_Y_OFFSET used after probeing

Kulitorum, 2017-01-14:
When I execute a home all command, the Z_PROBE_X_OFFSET is used after probeing.

My min_x_pos is -33 min_y_pos is -8
My Z_PROBE_X_OFFSET is 33, Z_PROBE_Y_OFFSET is -8
EXTRUDER_IS_Z_PROBE = 0 (not sure what this is for)

I know that the numbers are the same, this is a coincident.
Probe is a BLTOUCH
I use the probe to do homing on z axis and also for autoleveling.

When I run a G28 command, this happens:

Home X
Home Y
Home Z (and then lift the head correctly)
Move to 33,8
Move to -33,-8 (min XY)

I traced the code, everything looks correct, except that the PrintLine::moveRelativeDistanceInSteps command in Printer::startProbing being executed after the probeing has finished.

I cant figure out why, but aparrently somehow the probe commands takes precedense?

I tried after the comment "//Printer::offsetZ = 0; // we correct this with probe height" to add 

xExtra = 150; yExtra = 150;

And this clearly demonstrates the issue.

Thank you.

repetier, 2017-01-14:
Have you homing order XYPreheatZ? What are your probing coordinates?

What does
Home Z (and then lift the head correctly)
Move to 33,8
Move to -33,-8 (min XY)

mean exactly? I mean home z has so many moves and with special homing coordinates as required with z probe for min I can not see where what is going wrong from description and I have no printer with such a setup. I have one with 0 offset where extruder is z probe, but that would not show the error since offset is always 0.

Kulitorum, 2017-01-14:
It means that the printer runs the homeing z using the probe (at a wrong location) and then moves to 33,8 and then moves to -33,-8

https://www.youtube.com/watch?v=NUb0aVI02fU

On the BLTOUCH, the red light is on when the probe is UP. Its moved down using the "Probe start script" and up using the "Probe finished script"

Kulitorum, 2017-01-14:
Heres what I tried:

Change min-x and min-y to 0,0 and probe_x_offset, probe_y_offset to 0,0
This makes the printer probe at 0,0 - where it rests on the endstops.

Now setting minx and miny to -33,-8 makes the printer probe at 0,0

So I set probe-offset-x to 24 makes the printer probe at 0,0 - then move to x24 and then back to x0

So - as I said - the code to make sure the probe is inside the print area is getting run after the probing has happend.

Blue-Marlin, 2017-01-14:
One of the problems of BLTouch is its behaviour to signal a touch only for 5ms - that's easy to miss.
I don't know how probes are handled in Repetier, but Marlin has problems to detect the pulse when probing speed is too slow.

Kulitorum, 2017-01-17:
No problem with BLTOUCH in repetier.

repetier, 2017-01-23:
Had a similar problem with measuring. Reason was that target coordinates were outside xy min/max range and move was only executed partly. New version now issues a error and stops executing gcodes until 999 when that happens. You might need to change homing to XY Temperature Z with temp = 0. This allows setting z homing position to a point where it is allowed to activate.

apphead, 2017-05-25:
`You might need to change homing to XY Temperature Z with temp = 0. This allows setting z homing position to a point where it is allowed to activate.`
Hahaha, that did the trick for me. XY would not go to the ZHOME_X_POS and ZHOME_Y_POS on homing with HOME_ORDER_XYZ.

Z-Probing with a cartesian and BLTouch unfortunately remains a major PITA, ever since I tried to use Repetier. Pitty. Also the documentation is badly structred (as in in-structions), lacking visualisation of offsets etc and the naming of the variables is chosen to cause confusion: Z_PROBE_Z_OFFSET, Z_PROBE_BED_DISTANCE, Z_PROBE_HEIGHT (=Z_PROBE_Z_OFFSET)
The numerous posts as well as the dozens of people who haven given up Repetier because of the ABL stuff is proof enough that the whole topic should be addressed properly.