repetier/RepetierFirmware #628: Ramps fd v2 heated bed and hotend heater on wrong pin

jds598, 2017-01-17:
I just finished building a ramps fd v2.1 and couldn't get the heated bed or the hotend working. The heater pins are reversed for the two in pins.h

fullTalgoRythm, 2017-02-07:
Are you sure that's right surely just update in configuration.h

```diff
// Which pin enables the heater
-#define EXT0_HEATER_PIN HEATER_0_PIN
+#define EXT0_HEATER_PIN HEATER_1_PIN

 /** \brief Pin to enable heater for bed. */
-#define HEATED_BED_HEATER_PIN HEATER_1_PIN
+#define HEATED_BED_HEATER_PIN HEATER_0_PIN
```

jamesarm97, 2017-02-07:
According to the Ramps-FD v2 schematic they are using (or at least the schematic is labeled) D8 for Heatbed and D9 for Extruder1. I just checked my old Repetier Ramps-FD v2 source code I compiled for my board over a year ago and in Configuration.h I do have EXT0_HEATER_PIN HEAT_1_PIN

repetier, 2017-02-07:
Sound like it was always wrongly ordered since I got the pin definition. But changing it now is also bad as it would make all existing configurations invalid. So I guess I make a remark in pins.h for next update so users see it.

fullTalgoRythm, 2017-02-08:
As far as I can tell pins.h is correct for boards  403/404

```C
#define HEATER_0_PIN     8
#define HEATER_1_PIN     9 
```

Are you saying you'd normally reverse those round to match the defaults in configuration.h?

@jamesarm97 if you already reversed your configuration.h you can always try the following at the bottom of configuration.h

```C
#define STRVAL(x) #x
#define VAL(x) STRVAL(x)

#pragma message "HEATER PIN: [0]=" VAL(HEATER_0_PIN) " [1]=" VAL(HEATER_1_PIN)
#pragma message "HEATBED HEATER PIN: " VAL(HEATED_BED_HEATER_PIN)
#pragma message "EXT [0] HEATER PIN: " VAL(EXT0_HEATER_PIN)
```
Output is a bit messy but with the swap of EXT0 / HEATBED in configuration.h this gives me

```
sketch/Configuration.h:1763:74: note: #pragma message: HEATER PIN: [0]=8 [1]=9                                                          
sketch/Configuration.h:1765:65: note: #pragma message: HEATBED HEATER PIN: 8
sketch/Configuration.h:1764:57: note: #pragma message: EXT [0] HEATER PIN: 9                                     
```