repetier/RepetierFirmware #630: Ignore fan M106 command missing from recent UI menus in dev

JBKingdon, 2017-01-21:
Hi,
I suspect this is just a point in time while things are getting changed around, but recent development builds seem to be missing the ignore M106 menu option when using full graphic LCD during print from SD. I rely on that quite a bit, so please can we have it back :)

repetier, 2017-01-22:
It is now in setup menu and not gone:-)

closed 2017-01-22 by repetier

JBKingdon, 2017-01-22:
You can't get to the setup menu during a print, which is when I need the ignore M106 command. It needs to go on the new "while printing" menu, beside the option to set a fan speed.

JBKingdon, 2017-01-22:
I put a copy back in the main menu and it seems to work fine. My usual use-case is while printing I realise that I have completely inappropriate cooling settings and manually set a fan speed that won't be a disaster, and then set ignore M106 to avoid picking up the next bad value in the g-code.

What's the distinction between UI_MENU_ACTIONCOMMAND_FILTER and UI_MENU_ACTIONCOMMAND_FILTER_T? I wasn't quite sure which one to use, but the _T seemed to work ok.

repetier, 2017-01-23:
For next update I have added it to main menu after fan speeds.

See doc/menustructure.txt for more infos how to make your menu and what the suffixes do. _T is for translated text, means you enter translation id instead of text.

JBKingdon, 2017-01-23:
Many thanks. Dev is starting to look really good now - great work!