repetier/RepetierFirmware #631: DELTA_DIAGONAL_CORRECTION_A/B/C

tcurdt, 2017-01-24:
I am currently trying to calibrate my delta based on this http://www.thingiverse.com/thing:1274733/

  https://www.repetier.com/firmware/v092/images/deltacolumnnames2.png

From the schematic it's not exactly clear what `DELTA_DIAGONAL_CORRECTION_A/B/C` is. I assume it's a correction to the DELTA_DIAGONAL_ROD to be able correct for the individuals lengths of the rods. Is that correct?

Would be nice to have this clarified/included in the schematic.

repetier, 2017-01-24:
Yes, that is correct. Value gets simply added to diagonal rod length for computations.

tcurdt, 2017-01-24:
Awesome. Thanks for the quick response.
If I find the time - would you accept a contribution for a better schematic?

tcurdt, 2017-01-24:
For reference I am struggling with my calibration http://www.thingiverse.com/thing:1274733/#comments

A print with

```
DELTA_DIAGONAL_ROD 218
DELTA_DIAGONAL_CORRECTION_A 0
DELTA_DIAGONAL_CORRECTION_B 0
DELTA_DIAGONAL_CORRECTION_C 0
```
has pretty much the same size as a print with

```
DELTA_DIAGONAL_ROD 217.18
DELTA_DIAGONAL_CORRECTION_A -0.07
DELTA_DIAGONAL_CORRECTION_B 0.09
DELTA_DIAGONAL_CORRECTION_C -0.03
```
My intention was enlarge the object by 2mm on the xy.

repetier, 2017-01-24:
You mean the picture in config tool? Sure if it is better I have no objections replacing it.

2mm bigger from what width? That is a big error already but remember that you need diagonal+horizontal radius fixed or you get a bulb on your bed. As you have the diagonals working in diagonals you must add more then 1mm more like 1.5mm i would guess. But that is a range you should be able to measure for diagonals. SO maybe the error is somewhere else. Some users have for example the effect that the effector rotates while moving. That is not what it should do in theory and no variable compensates rotation. So I hope it not that causing the difference.

tcurdt, 2017-01-24:
When I print a 100mm object that comes out as 98mm on the xy. The z is fine (but I haven't printed more than 10mm in that direction yet). Now I want to fix the settings that it actually comes out as 100mm.

So you are saying my corrections of the rods are (with currently just around 0.8mm) too small and the behaviour of geting pretty much the same size is expected?

TBH I was just following the referenced procedure on thingiverse. According to the article on thingiverse adjusting the rod lengths should be enough - but if you could point me into the right direction that would be great.

By measuring the diagonals you mean the rods?

repetier, 2017-01-24:
It is had to see with all these equation when what change show much, so don't take my comments too serious. Just my feelings and since my deltas are calibrated for years it is a good time back.

The thingiverse solution is iterative, so may need more then one update. You should also check if anything else gets worser especially how extruder moves over bed might get wrong.

diagonals = rods, yes.

tcurdt, 2017-01-24:
I don't have a caliper long enough but according to a ruler the rods should be all around 217.5. The first print with

```
DELTA_DIAGONAL_ROD 218
DELTA_DIAGONAL_CORRECTION_A 0
DELTA_DIAGONAL_CORRECTION_B 0
DELTA_DIAGONAL_CORRECTION_C 0
```

resulted in a length of pretty close to 98mm (for 100mm piece).

Apparently I used the formula from thingiverse the wrong way. Adjusting it to the correct usage it resulted in

```
DELTA_DIAGONAL_ROD 215.28
DELTA_DIAGONAL_CORRECTION_A -0.20
DELTA_DIAGONAL_CORRECTION_B 0.29
DELTA_DIAGONAL_CORRECTION_C -0.08
```

and I am still ending up at 98mm. Now I am starting to want to throw things *sigh*

How can this have so little or no impact?
Before you ask - I have the EEPROM disabled.

I tempted to do a print that is completely off just to see if it has any impact at all.

Any ideas, @repetier? I've run out.

andyhomes, 2017-01-24:
Hello.
You asked any idea...
I have one: are you sure you flashed FW successfully after changing settings?
There is no miracle. I think it's something so simple that could explain this.
Check everithing again: EEPROM_MODE is 0; printer settings with m205 and also that it reports in console on start etc

BTW if you have EEPOM it's very simple to tweak this parameters w/o rewriting FW with M206:
Just "print":
M206 T3 P881  X215.28  ; Diagonal rod length [mm]
M206 T3 P933  X-0.20    ; Corr. diagonal A [mm]
M206 T3 P937  X0.29    ; Corr. diagonal B [mm]
M206 T3 P941  X-0.08    ; Corr. diagonal C [mm]

tcurdt, 2017-01-24:
@andyhomes it always says upload done and the board restarts.

`M205` gives `Error:No EEPROM support compiled.`

tcurdt, 2017-01-24:
Just to make sure I am not going insane I did a print with:

```
DELTA_DIAGONAL_ROD 210
DELTA_DIAGONAL_CORRECTION_A 0
DELTA_DIAGONAL_CORRECTION_B 0
DELTA_DIAGONAL_CORRECTION_C 0
```

and finally I see a difference in size. It's almost spot on by just luck.

That said I still would love understand this. The real rod length is about `217.5`. How come this has to be such a big difference. (Almost 8mm!?) There must be an error somewhere else.

andyhomes, 2017-01-24:
And also check 
END_EFFECTOR_HORIZONTAL_OFFSET
CARRIAGE_HORIZONTAL_OFFSET

And you probably know (in Repetier.h and from explanations in the config file):
#define ROD_RADIUS (PRINTER_RADIUS-END_EFFECTOR_HORIZONTAL_OFFSET-CARRIAGE_HORIZONTAL_OFFSET)

tcurdt, 2017-01-24:
Hm, we might be on to something here, @andyhomes.
It's a Kossel mini and it came with this setting:

```
#define ROD_RADIUS 112
#define PRINTER_RADIUS 112
#define END_EFFECTOR_HORIZONTAL_OFFSET 0
#define CARRIAGE_HORIZONTAL_OFFSET 0
```

but when I look at the difference of `ROD_RADIUS` and `PRINTER_RADIUS` that can't be right.

repetier, 2017-01-25:
Rod length and horizontal radius have influence on positioning. I wonder that they only change diagonal length, but have as said no experience what changes real size.

tcurdt, 2017-01-25:
@repetier but would you agree that the `ROD_RADIUS` and `PRINTER_RADIUS` being equal looks wrong?

tcurdt, 2017-01-26:
I measured and got
```
ROD_RADIUS 112
PRINTER_RADIUS 130
```
but when printing that made no whatsoever difference :-(

andyhomes, 2017-01-27:
PRINTER_RADIUS is only configuration parameter that is used to determine ROD_RADIUS (in Repetier.h, unless it's not defined somewhere else) that's used everywhere for calculations.
ROD_RADIUS = PRINTER_RADIUS - END_EFFECTOR_HORIZONTAL_OFFSET - CARRIAGE_HORIZONTAL_OFFSET (these offsets are only to define ROD_RADIUS too, again unless the ROD_RADIUS is already defined)
So, `112 = 112 - 0 - 0` is ok.

Which version of FW do you use?
I don't see any obstacles in Repetier's original dev/tag branches to have difference in print results changing these parameters, so maybe something in code base that you use?

tcurdt, 2017-01-27:
I am using [0.92](https://www.repetier.com/firmware/v092/). I've uploaded the original configuration and made the adjustments, then compared the configurations with a diff.

According to [this sketch](https://www.repetier.com/firmware/v092/images/deltacolumnnames2.png) the ROD_RADIUS is the distance from the outer joint to the inner joint. And the PRINTER_RADIUS is from the nozzle to the outer joint. Given that the effector width is of course not 0 they should not be the same.

At least that's how I understand the sketch.

ProfDrYoMan, 2017-03-27:
Hello,
first of all, I suggest to use the dev version, because there are many fixes especially for deltas.

Then you should try to measure the length of the diagonal rods by hand as precise as possible. The best is to build them exactly the same.

At last you should try http://escher3d.com/pages/wizards/wizarddelta.php with 6 factors and 10 measurements. Anything but the diagonal rod length will be corrected.

I am down to < 0,05 mm error over the whole board with 230 mm diameter!