repetier/RepetierFirmware #635: Excessive flow of plastic in the corners

Sanitar777, 2017-02-10:
Hello Repetier and All!

Repetier, thanks for your work! Nice firmware.

I have: 
Printer CoreXY + Arduino DUE + RADDS + Dual Z-drivers + Dual Z-Endstop at the MAX position and Repatier-Firmware DEV of latest revision. 

During printing, the extruder squeezes out the excess plastic on the corners of the model and the convex angles are obtained. See pictures below.

So does the slicer, there are gaps on the corners:
[image]

So does the printer, the excess plastic:
[image]

And so it turns the angles on the model:
[image]

My Configuration.h file:
[Configuration.zip](https://github.com/repetier/Repetier-Firmware/files/767281/Configuration.zip)

Please help me to rectify the situation. 
What and where should be changed in the firmware so that the corners were smoother? 
I tried different settings, nothing helps. 

Would be grateful for your help in this problem.
Thank you.

repetier, 2017-02-10:
The problem is back pressure when you get slower in the corners. While extruder starts extruding slower due to speed going down in corners,  there is pressure from past move that makes the filament extrude faster then the motor pushes. When you accelerate this inverses. The effect gets even more visible on bowden type extruders, since these have a long "spring" storing the pressure. So what you want are small differences in speed.  So here is what you can do:
- Increasing jerk helps to not slow down as much, but that makes the printer shake and at some point you will loose steps.
- Print slower. This also reduces difference in corners.
- Acceleration. Not so sure what is better. Slow acceleration gives you more time to adjust to new speed while fast acceleration reduces areas of slow speed.
- Use advance algorithm. This moves extruder even slower when decelerating and faster when accelerating, so it works the effect. Problem is there is a limit what you can undo that way. Especially with bowden systems you have to undo so much that the extruder would need to retract during slowdown. So extruder works very hard on your filament and that causes new problems like grinding your filament. So it can help but do not too high with correction. 

As a simple test print with jerk 25 and max. speed 25. You should see the effect is much less prominent and maybe even gone.

Sanitar777, 2017-02-10:
Thank you so much for a very quick detailed response. 

Sorry, I didn't say that I use a direct extruder and print speed of 80mm/s and more. 
Speed 25mm/s I'm not interested in such a small speed prints all good. Sorry. 
What can I do with this problem at high speeds?

repetier, 2017-02-10:
I was not assuming you want to print 25mm/s. That is a test to see if the problem then vanishes so you see that this is the problem you currently have.

Using the 3 solutions in combination you can improve it. One option is always to reduce the speed for outer perimeter to a value where the effect is tolerable. And with direct drive extruder advance is much more effective. This is the visible one and errors there are what you see most. Infill - who cares about this effect. It might even be good adding some extra bonding to perimeter when turning direction.

Sanitar777, 2017-02-10:
Of course, I always do the speed at the outer perimeter is much smaller than the rest.
But I don't understand how it works ADVANCED_L and ADVANCED_K. Also I don't understand which values to use for these parameters and how to use them.

repetier, 2017-02-11:
Use only L term. That is linear while K is nonlinear factor getting too fast easily. L=0 is no advance, and positive value is what you need. 10-60 is a typical range I think. Bigger = more correction. As I said I would not try to go that high that direction reverses, but maybe even that works and is required. Here faster acceleration needs higher values as we have less time to correct. So these can also work together for optimum quality.

Sanitar777, 2017-02-11:
Thank you very much. 
I will try to experiment with setting the value of L more. 
Before I tried to change the values of L, but in the range of 5-15.

closed 2017-02-15 by Sanitar777

Sanitar777, 2017-07-11:
I forgot to write a report about my research.
Here is the result. ADVANCED_L = 55
[image]
Thank you!