repetier/RepetierFirmware #637: Firmware keeps sending "wait" after target temperature is reached

Kulitorum, 2017-02-16:
In the latest version of the dev branch, all the three printers that I am running that version on, they never get past the loop waiting for bed to heat up.

So I start a print (Simplify3D) and the bed heats up. When the temp is reached, the firmware keeps sending "wait" and I have to "Force Next" to start printing.
[Wait.zip](https://github.com/repetier/Repetier-Firmware/files/780243/Wait.zip)

Attached example gcode and config file.
[Wait.zip](https://github.com/repetier/Repetier-Firmware/files/780244/Wait.zip)

repetier, 2017-02-16:
What does the log look like? "wait" just means host is not sending lines also firmware is ready to accept them. The ok comes normally directly after parsing it. Do you have updated Simplify3D and it does that handle differently now?

Kulitorum, 2017-02-17:
Well, the host has the temperature of the bed at 59, while the display on the printer is at 60, so it seems that maybe when the printer reaches target, it stops sending data to the host.

So all the host is getting is "Wait" \n "wait" while the printer display shows the temp of the bed at 60c, the host shows the bed temp as 59c.

  Connected to machine!
READ: Info:External Reset
READ: Detected EEPROM version:18
READ: Info:Autoleveling enabled
READ: Transformation matrix: 0.999998 0.000000 0.002137 0.000006 0.999996 -0.002660 -0.002137 0.002660 0.999994
SENT: T0
SENT: M105
READ: Free RAM:3487
READ: start
READ: Detected EEPROM version:18
READ: Info:Autoleveling enabled
READ: Transformation matrix: 0.999998 0.000000 0.002137 0.000006 0.999996 -0.002660 -0.002137 0.002660 0.999994
SENT: M105
READ: Free RAM:3487
SENT: M105
READ: SelectExtruder:0
READ: FlowMultiply:100
READ: ok 0
READ: T:0.00 /0 B:0.00 /0 B@:0 @:0
READ: ok 0
READ: T:13.16 /0 B:19.85 /0 B@:0 @:0
READ: wait
SENT: M105
READ: ok 0
READ: T:12.98 /0 B:19.70 /0 B@:0 @:0
SENT: M140 S60
READ: ok 0
READ: TargetBed:60
READ: wait
SENT: M105
READ: ok 0
READ: T:12.81 /0 B:20.75 /60 B@:255 @:0
READ: wait
SENT: M105
READ: wait
READ: ok 0
READ: T:12.98 /0 B:21.95 /60 B@:255 @:0
READ: wait
SENT: M105
READ: ok 0
READ: T:12.98 /0 B:22.55 /60 B@:255 @:0
READ: wait
SENT: M105
READ: ok 0
READ: T:12.98 /0 B:23.25 /60 B@:255 @:0
READ: wait
READ: wait
SENT: M105
READ: ok 0
READ: T:12.98 /0 B:24.00 /60 B@:255 @:0
READ: wait
SENT: M105
READ: ok 0
READ: T:12.98 /0 B:24.62 /60 B@:255 @:0
READ: wait
SENT: M105
READ: wait
READ: ok 0
READ: T:12.98 /0 B:25.00 /60 B@:255 @:0
READ: wait
SENT: M105
READ: ok 0
READ: T:12.98 /0 B:25.50 /60 B@:255 @:0
READ: wait
SENT: M105
READ: ok 0
[... 587 lines cut ...]
READ: wait
READ: wait
READ: wait
READ: wait
READ: wait
READ: wait
READ: wait
READ: wait
READ: wait
READ: wait
READ: wait
READ: wait
READ: wait
READ: wait
READ: wait
READ: wait
READ: wait
READ: wait
READ: wait
READ: wait
READ: wait
READ: wait
READ: wait
READ: wait
READ: wait

repetier, 2017-02-17:
No error in firmware. See this critical part:
READ: wait
SENT: G90
SENT: M82
SENT: M106 S0
SENT: M140 S60
SENT: M190 S60
READ: ok 0
SENT: M105
READ: ok 0
READ: ok 0
READ: ok 0
READ: ok 0
READ: ok 0
READ: T:13.68 /0 B:59.95 /60 B@:255 @:0

You see 6 commands and they get 6 times "ok" so all send lines were acknowleged. So it is on simplify to send the next command which simply does not happen.
You also see they are not using line numbers and probably also no checksums or they did delete that from log. Try sending same file with repetier server or host and I think it will work.