repetier/RepetierFirmware #638: G32 is bypassing the previous command

thiagopeixoto16, 2017-02-19:
I have a Cartesian machine where z-min endstop is z-probe mounted on a servo.
I have the following start code on my slicer:
G28 Y0 ; home X and Y axis
G28 X0 ; home X and Y axis
G1 X100 Y100 F5400;
G28 Z0 
G1 Z20 
G32 S2 ; Auto bed correction
However the machine doesn't raises z by 20 after the **G28 Z** nas is somehow jumping to G32 tha makes my servo to crash with the bed...
I tried to put the G1 Z20 into the Z_PROBE_START_SCRIPT but I had the same servo crash...
The only I could workaround the problem is by inserting a wait of 50ms like below:
G28 Y0 ; home X and Y axis
G28 X0 ; home X and Y axis
G1 X100 Y100 F5400;
G28 Z0 
G1 Z20 
**G4 P50**
G32 S2 ; Auto bed correction
What I'm doing wrong?

repetier, 2017-02-19:
First use dev version. There we enhanced such things to a good extend. You could use homing procedure XY Preheat Z which allows setting probing position for Z. That allows you to simply use G28 for homing.

Skip S2 in G32 if you do it always (which is normally not necessary). This stores to eeprom reducing eeprom lifetime. Ok you can do many writes but why writing if not necessary. So no panic.

In dev versionI think you could do G32 without any extra commands. It goes to z = RMath::max(EEPROM::zProbeBedDistance() + (EEPROM::zProbeHeight() > 0 ? EEPROM::zProbeHeight() : 0) so setting z probe bed distance 20 should suffice to ensure correct height.

In older firmware versions I guess it started your script and since G1 Z20 was just buffered it had no time to go up. So M400 before servo commands would have protected the servo in start scripts here.

thiagopeixoto16, 2017-02-19:
Thanks for the quickly reply.
Yes, I'm on dev, sorry I forgot to mention. 1.0.0.dev from Feb 8th.
Do you think that that the "G1 Z20" bypass thing is a bug?