repetier/RepetierFirmware #641: Printer::startProbing issue with negative XY values on Deltas?

fitzterra, 2017-02-27:
I've recently switched to the **development** branch where I keep getting the error "**Could not activate z-probe offset due to coordinate constraints**" when trying most commands involving a z-probe. This is on a home built Delta bot.

From `Printer::startProbing` : 
```C++
	// Fix position to be inside print area when probe is enabled
#if EXTRUDER_IS_Z_PROBE == 0	
	float ZPOffsetX = EEPROM::zProbeXOffset();
	float ZPOffsetY = EEPROM::zProbeYOffset();
	if((ZPOffsetX > 0 && Printer::currentPosition[X_AXIS] - ZPOffsetX < Printer::xMin) ||
	   (ZPOffsetY > 0 && Printer::currentPosition[Y_AXIS] - ZPOffsetY < Printer::yMin) ||
	   (ZPOffsetX < 0 && Printer::currentPosition[X_AXIS] - ZPOffsetX > Printer::xMin + Printer::xLength) ||
   	   (ZPOffsetY < 0 && Printer::currentPosition[Y_AXIS] - ZPOffsetY > Printer::yMin + Printer::yLength)) {
        GCode::fatalError(PSTR("Could not activate z-probe offset due to coordinate constraints - result is inprecise!"));
        return false;
    } else {
	    // Update position	
        Printer::offsetX = -ZPOffsetX;
        Printer::offsetY = -ZPOffsetY;
        Printer::offsetZ = 0;    
#if FEATURE_AUTOLEVEL
        // we must not change z for the probe offset even if we are rotated, so add a correction for z
        float dx,dy;
        transformToPrinter(EEPROM::zProbeXOffset(),EEPROM::zProbeYOffset(),0,dx,dy,offsetZ2);
        //Com::printFLN(PSTR("ZPOffset2:"),offsetZ2,3);
#endif    
    }
#endif	
 ```

For me `xMin` and `yMin` are both zero and anytime the probe position is to the negative side of 0,0 (center of build platform), the above test is true resulting in the error.

I'm not sure if on a Delta `xMin` and `yMin` should be zero or if it should be the negative of the build radius.

If it should be zero as it is in my case, then there may be a problem in the code above for deltas.

If `xMin` and `yMin` should *not* be zero, then I may have a problem on my config somewhere in which case I would appreciate some insights into where I can look to fix this.

Any help would be appreciated.

PS - a great big thank you for the effort and hard work in creating and sharing this awesome code. It's highly appreciated!

repetier, 2017-02-28:
I have changed that lately to create a error message instead of doing wrong stuff. Still wonder why it worked before as xMin, yMin is 0 and makes no sense for deltas. Have modified the tests now to have a special condition that should work. Could at least probe at negative xy coordinates with my delta. So I mark this as fixed unless you report that the update still fails.

fitzterra, 2017-03-04:
Thanks @repetier - issue solved except for the spelling mistake which I have commented on inline in the commit diff here : 92b40ea0cca4ab468f84ab4257e4f44c41fa240e

closed 2017-03-04 by fitzterra

repetier, 2017-03-04:
DEL spelling error is also fixed since yesterday. But thanks for the hint.