repetier/RepetierFirmware #642: 0.92.8 not released als released file and not tagged

ProfDrYoMan, 2017-03-01:
Hi,
Since 0.92.8 seems to be the last stable release it should be tagged and marked as release?

Sidequestion:
I start using a delta printer (anycubic / kossel). What version should I go for? Dev branch?

repetier, 2017-03-01:
0.92.9 is latest stable but I would use dev version for the improved and corrected delta behaviours especially regarding probing. Also called dev it is stable.

Tags - ok I forget them from time to time and now I'm past it and do not know where it would need to be inserted. Will try to think for next releases to tag them.

closed 2017-03-01 by repetier

ProfDrYoMan, 2017-03-01:
Ok.
If I understand you correctly you suggest to use this:
https://www.repetier.com/firmware/dev/index.php
?

repetier, 2017-03-01:
Exactly, that is what I suggest.

ProfDrYoMan, 2017-03-01:
Ok. Thank you.