repetier/RepetierFirmware #643: Layer shift

stecor7001, 2017-03-06:
Thank you for your efforts. I installed Repetier yesterday, and it was a breeze to setup on my machine - a Sharebot ng.

The print quality comes out better than Marlin's, i.e. The outer/inner perimeters are much more defined, and the printed surfaces look almost flawless.

However, there is a but - on one occasion, a small test part containing two stacked hexagonal prisms and a cylinder on top, I had layer shifting on the cylinder part, on the Y axis only. I never experienced shifting under Marlin with this machine, however thedrivers run @63-65C so I thought let's decrease the current a bit. The issue seemed resolved (i.e. No shifting any longer on the test part)so I went on to print something larger. Layer shift occured about 1hr in the print(as compared to minutes for the test part), and was only on Y. I replaced the stepper driver on Y and restarted the print. Another layer shift @ about 1h:20', however this time it was mostly on Y but X also missed a few steps.

Reverted back to Marlin (no other changes made to the printer) and it all worked fine(total print time 8hr45'.). This points in my opinion to a potential issue with the firmware. Same g-code, same settings. Latest version oh RF - dev branch.

The issue seems to occur while (or just after) a circle outline is performed.

Thank you for looking into it.

vinmicgra, 2017-03-06:
I've noticed a few reports of this issue in the last time. 
Could you please check latest dev branch with following manual addition to configuration.h:

#define REDUCE_ON_SMALL_SEGMENTS

repetier, 2017-03-06:
In addition also if shift is always away from y endstop set ALWAYS_CHECK_ENDSTOPS 0. Crosstalk often is also cause of shifts.
Last thing is reduce xy jerk in eeprom. Just the typical reasons for shift.

stecor7001, 2017-03-07:
Thank you both for responding. 

Vinmicgra: I thought Reducexxx is meant to compensate for backlash - which is an issue I don't have. It also seems to add overhead as additional moves are generated. It would also add additional vibrations I would think (accel vector is now full blast vs the sideways accel when switching between circle segments). 
Would you kindly help me understand why I would want to enable this option? 

Repetier: Okay, makes sense that crosstalk may arise, even though the signal path is definitely low impedance on the endstops - mine go to +5V when not triggered through a 220ohm resistor and to ground when on. 

Jerk is exacly the same as in Marlin - 20 for XY, so would hope it is not a problem.

I will try the endstop change mentioned above and let you know. This might take a day though.

Thanks both.

repetier, 2017-03-07:
#define REDUCE_ON_SMALL_SEGMENTS
Has nothing to do with backlash and can really help