repetier/RepetierFirmware #648: List G-codes on command or autobuild G-code list via e.b. doxygen

ProfDrYoMan, 2017-03-17:
Hello,

yes, I know, flash space is not endless. But this would be nice to get complete G-code informations. It could be a configuration.h option.
http://reprap.org/wiki/G-code#G:_List_all_G-codes
Maybe it would be nice get a documentation for a specific G-code on this syntax:
Input:
G?1
Output:
1 : Linear Move 

Usage
    G0 Xnnn Ynnn Znnn Ennn Fnnn Snnn 
    G1 Xnnn Ynnn Znnn Ennn Fnnn Snnn 
Parameters 
....

Another solution would be to write doxygen documentation in the G-code parser for every G-code and let doxygen generate the documentation.

I am new to repetier and for me the hardest part is to find and understand functionality, because it is often barely or even not actually right documented.

repetier, 2017-03-17:
Have you already looked into repetier.ino where all gcodes in repetier should be documented in a short way.

ProfDrYoMan, 2017-03-17:
I did. But are they actually all there and all correct.
Embedding the documentation into the code would help keeping them correct.

repetier, 2017-03-21:
Sprading docs is too complicated for most, taht is why I concentrate them in ino file. Try to update them everytime I make changes, but bugs are still possible.

closed 2017-03-21 by repetier