repetier/RepetierFirmware #658: Request for support of UC1701 Mini12864 LCD

GarrethX, 2017-04-03:
I've been fighting with the UC1701 Mini12864 LCD at Arduino DUE for some time, and I couldn't get it working in repetier. In Marlin it was working ok, but maybe you can support me because maybe I'm missing something there.

I use due3dom mini board, and it works on marlin and also in examples from u8glib on software SPI with the constructor:
U8GLIB_MINI12864 u8g(76,75,45,46,U8G_PIN_NONE); where:
76 - SCK
75 - MOSI
45 - CS
46 - A0

The rest of the pins (encoder, buttons, SD_CS, reset) are as in the standard full_graphic_discount_controller.

With the heavy optimization of u8glib in repetier I'm not able to get deeper to what is going wrong. HW SPI I can't get to work at all with the standard u8glib, but even having only SW SPI would be great if this would not collide with SD support of course. 

Thanks!

Kulitorum, 2017-05-13:
Is this the same as MKS Mini 12864LCD ? - Asking because I was hoping for support for that :)

repetier, 2017-05-13:
I have just added the driver for that board. It works configuration wise like the ST920 in controller 11 but you need to define U8GLIB_MINI12864_2X_SW_SPI instead to use that driver. Having no board and such display it is now available but only in uiconfig.h for those configuring it explicitly.

GarrethX, 2017-05-13:
@Kulitorum - yes, it's that display
@repetier - Thank you very much, you're the best!:) I'll try to find time next weekend to test it out. On Marlin I managed to add this myself to use SW SPI and it works, but on Repetier I failed with this. If it will not work, would you be interested if I donated you the money for that display so you could test it? It's about $13 on Aliexpress.

repetier, 2017-05-13:
It should work. If you need the 1x driver that is also included but no define to select it.
I have an other tester already that also wants to configure the display and I already tested that it compiles, so I hope no display is needed.

Bob-the-Kuhn, 2017-05-25:
Can you provide details on how you setup Marlin to work with this display?

repetier, 2017-05-25:
Seems like it requires u8glib 2 to get this display working, also it is listed in u8glib as well :-(

GarrethX, 2017-05-25:
In the evening I'll send you guys how I did it. It required a little bit of
soldering as I gave up with fiddling in the software.
I discovered that I had a Chip select conflict with SD card reader, so I
spliced the data lines from the display (unsoldered two goldpins which
connect display data lines to the board and replaced them by angled
goldpins), then I connected the display data lines to free arduino outputs
and set a software spi connection on those pins.

SD card remained at HW spi, and works without problems.

I'll post my hand drawn schematic and marlin for reference, but when I'm
back at home.

GarrethX, 2017-05-25:
Below what I promised. The schematic is a little bit vague and not contain the power supply traces, but you should get the idea. On the red I marked the modification which I did, depending on your board it may be different. 

Also attached marlin for due with my config for that display. The pins I modified are for the board Due3dom mini , so there search for the presence of mini12864 and adapt to your board.

I don't know why the same method is not working for repetier. Somewhere alone the software SPI implementation in repetier must be probably some optimization which is just "too much", but I couldn't trace it. Maybe it's too fast, maybe it's too slow, no idea...

[image]

[Marlin-Duo.zip](https://github.com/repetier/Repetier-Firmware/files/1029114/Marlin-Duo.zip)

Bob-the-Kuhn, 2017-05-25:
Many thanks for the details.

Modifying the LCD - beyond the capabilities of most users but not all.

Is this the MKS MINI12864 ?

GarrethX, 2017-05-25:
We all have limitations:) LCD works great with u8glib2 and HWSPI, but it was beyond my capabilities to modify either repetier or marlin to use that library. I tried, but failed badly... This modification is not destructive for the board it self, so it is possible to go back (unless you destroy something in the process), it may be tricky to remove these two pins, but good thin cutting pliers and it takes 15 minutes. 

Yea, it's MKS mini12864, but not the OLED version which sometimes can be found. It's this: https://www.banggood.com/MKS-Mini-12864LCD-Controller-For-3D-Printer-Marlin-Stand-Inserted-SD-Card-p-1092416.html (I'm not sure how long before the link expires).

Bob-the-Kuhn, 2017-05-25:
What LCD did you select in Configuration.h?

GarrethX, 2017-05-25:
#define MINIPANEL

But you must go through the firmware I posted, in file ultralcd_impl_DOGM.h I modified line 176:
```
#elif ENABLED(MINIPANEL)
  // The MINIPanel display
  //U8GLIB_MINI12864 u8g(DOGLCD_CS, DOGLCD_A0);  // 8 stripes
  U8GLIB_MINI12864_2X u8g(53,51,DOGLCD_CS, DOGLCD_A0,U8G_PIN_NONE);
```

Bob-the-Kuhn, 2017-05-25:
This is looking encouraging.  😊

samezrp, 2017-11-08:
Hi guys,
I would like to refresh the topic as on the Polish Fabrykator we (@Garreth and @dziobu) did some research and finally enabled MKS MINI 12864 display to work with Arduino Due in HW SPI mode. See attached file.
[ArduinoDue HW SPI for uc1701_mini12864.zip](https://github.com/repetier/Repetier-Firmware/files/1455389/ArduinoDue.HW.SPI.for.uc1701_mini12864.zip)
Reason for not working was Arduino speed but the real problem was U8Glib itself.
Theoretically solution for too high speed is slow down SPI with SPI divider, unfortunately we found that this feature does not work for U8Glib.
With a very valuable help of user @dziobu we found that the reason is the way how U8Glib starts transmission. The library backups SPI environment, sets it to its own settings and restores after transmission disregarding any SPI settings done in code. In attached file above you can find extract from main library corrected to support in a proper way MKS MINI 12864 display.
Setting of SPI environment is done in lines 86-88:
```
#define _LCD_SPI_CLOCK_HZ 4000000
// tryb SPI: 0..3
#define _LCD_SPI_MODE 1
```
MODE 0 is working as well.
Would you mind taking a look into this code and include it in Repetier DEV releases, please?

erik23de, 2019-03-08:
This all is two years ago now. Firmware has changed since then. Are there now easier ways to get that display working? Anyone found a good doc for doing so?

Thank you all--