repetier/RepetierFirmware #661: Printer does z-hops during printing

HenryLoenwind, 2017-04-20:
Version: 185b05c08090a378415e03bef5f0d9f4a251d116
[Configuration_h.txt](https://github.com/repetier/Repetier-Firmware/files/945490/Configuration_h.txt)
[gcode.txt](https://github.com/repetier/Repetier-Firmware/files/945507/gcode.txt)

The printer does z-hops, up and down again, during printing. I checked the gcode and I don't see any z-changes within the layer. The hops seem to be where there's a small movement without extraction in the gcode.

Video: https://youtu.be/Xcf4hd5F2Rg

gcode visualization: [image]

HenryLoenwind, 2017-04-20:
Update:

After a bit of experimenting I can blame this on extruder-only `G1`'s with `DISTORTION_CORRECTION` enabled and active (`Printer::destinationSteps[Z_AXIS] < Printer::distortion.zMaxSteps()`).

repetier, 2017-04-21:
So the hopping is just a result of distortion map having bumps stored? Then we can close the issue?

HenryLoenwind, 2017-04-21:
No, there are no micro-bumps stored at exactly the locations my models have retractions. ;)

I'd say that PrintLine::queueCartesianMove()/queueCartesianSegmentTo() doesn't handle G1s with E but no XYZ correctly. It seems to get a different Z correction for them than for G1s with XYZ, resulting in a retraction G1 being executed as a move up. Example:

```
G1 X115.296 Y161.692 E155.5052
G1 F4800 E149.5052
G1 X114.082 Y161.135 F10800
G1 F4800 E155.5052
G1 X112.736 Y160.908 E155.5733 F1800
```

comes in and gets executed somewhat like:

```
x115.296 y161.692 e155.5052 z(0.3+0.12)
x115.296 y161.692 e149.5052 z(0.3+2.8)
x114.082 y161.135 e149.5052 z(0.3+0.12)
x114.082 y161.135 e155.5052 z(0.3+2.8)
x112.736 y160.908 e155.5733 z(0.3+0.12)
```

instead of

```
x115.296 y161.692 e155.5052 z(0.3+0.12)
x115.296 y161.692 e149.5052 z(0.3+0.12)
x114.082 y161.135 e149.5052 z(0.3+0.12)
x114.082 y161.135 e155.5052 z(0.3+0.12)
x112.736 y160.908 e155.5733 z(0.3+0.12)
```

I had a look at the code, but the distortion correction path is so different from the normal one that I cannot see anything there :(

repetier, 2017-04-21:
Are you using the dev version? In older there were big errors in distortion correcting resulting in wrong moves. If you are using dev version, does this also happen when distortion is disabled, so we can nail it down to the distortion special case?

HenryLoenwind, 2017-04-21:
Yes, dev. I linked the exact version I'm using as very first thing above.

And yes, this only happens when distortion correction is on and only below DISTORTION_END_HEIGHT. I *think* the hop distance gets smaller according to the distortion degrade, but that's from watching the printer with my eyes...

I also confirmed that there's nothing special about the model, it also happens with a simple hollow cube.

repetier, 2017-04-21:
What is your output of G33 L0 ?
I'm wondering why it is hopping and where it gets the hopping height from. If no xyz is given it should just use the last xyz.

You could compile with
#define DEBUG_DISTORTION
so all calls to
	void PrintLine::queueCartesianSegmentTo(uint8_t check_endstops, uint8_t pathOptimize) {
would write the z correction they assume. If you then run the short 5 lines it should log some more data which might give a hint what is going on here.

HenryLoenwind, 2017-04-21:
```
20:35:23.734: N19 G33 L0
20:35:23.738: G33 X51.00 Y11.00 Z0.057
20:35:23.739: G33 X51.00 Y50.83 Z0.317
20:35:23.742: G33 X51.00 Y90.66 Z0.425
20:35:23.743: G33 X51.00 Y130.49 Z0.435
20:35:23.746: G33 X51.00 Y170.32 Z0.367
20:35:23.748: G33 X51.00 Y210.15 Z0.202
20:35:23.750: G33 X51.00 Y249.98 Z-0.040
20:35:23.750: G33 X84.16 Y11.00 Z0.235
20:35:23.752: G33 X84.16 Y50.83 Z0.413
20:35:23.754: G33 X84.16 Y90.66 Z0.505
20:35:23.756: G33 X84.16 Y130.49 Z0.515
20:35:23.759: G33 X84.16 Y170.32 Z0.422
20:35:23.759: G33 X84.16 Y210.15 Z0.277
20:35:23.760: G33 X84.16 Y249.98 Z0.078
20:35:23.763: G33 X117.32 Y11.00 Z0.287
20:35:23.764: G33 X117.32 Y50.83 Z0.430
20:35:23.767: G33 X117.32 Y90.66 Z0.513
20:35:23.770: G33 X117.32 Y130.49 Z0.507
20:35:23.771: G33 X117.32 Y170.32 Z0.427
20:35:23.771: G33 X117.32 Y210.15 Z0.297
20:35:23.774: G33 X117.32 Y249.98 Z0.137
20:35:23.775: G33 X150.48 Y11.00 Z0.340
20:35:23.778: G33 X150.48 Y50.83 Z0.450
20:35:23.779: G33 X150.48 Y90.66 Z0.530
20:35:23.782: G33 X150.48 Y130.49 Z0.518
20:35:23.784: G33 X150.48 Y170.32 Z0.450
20:35:23.786: G33 X150.48 Y210.15 Z0.338
20:35:23.786: G33 X150.48 Y249.98 Z0.232
20:35:23.788: G33 X183.64 Y11.00 Z0.320
20:35:23.790: G33 X183.64 Y50.83 Z0.415
20:35:23.792: G33 X183.64 Y90.66 Z0.490
20:35:23.795: G33 X183.64 Y130.49 Z0.480
20:35:23.796: G33 X183.64 Y170.32 Z0.420
20:35:23.799: G33 X183.64 Y210.15 Z0.375
20:35:23.799: G33 X183.64 Y249.98 Z0.307
20:35:23.800: G33 X216.80 Y11.00 Z0.290
20:35:23.803: G33 X216.80 Y50.83 Z0.352
20:35:23.804: G33 X216.80 Y90.66 Z0.448
20:35:23.807: G33 X216.80 Y130.49 Z0.415
20:35:23.810: G33 X216.80 Y170.32 Z0.397
20:35:23.811: G33 X216.80 Y210.15 Z0.377
20:35:23.811: G33 X216.80 Y249.98 Z0.315
20:35:23.814: G33 X249.96 Y11.00 Z0.207
20:35:23.815: G33 X249.96 Y50.83 Z0.185
20:35:23.818: G33 X249.96 Y90.66 Z0.275
20:35:23.820: G33 X249.96 Y130.49 Z0.260
20:35:23.821: G33 X249.96 Y170.32 Z0.265
20:35:23.824: G33 X249.96 Y210.15 Z0.247
20:35:23.824: G33 X249.96 Y249.98 Z0.290
```

(Yes, I know the right side is nearly 0.3 high, but I want to get the software side working before I finetune everything...)

HenryLoenwind, 2017-04-21:
Ok, here 's the debug output.

A short one:

```
23:13:50.015 : N20 G1 E-1*99
23:13:50.020 : zCorr:0.402 atX:115.30 atY:161.69

23:14:12.829 : N21 G1 X115.296*69
23:14:12.834 : zCorr:0.422 atX:115.30 atY:161.69

23:14:19.162 : N22 G1 X115.296*70
23:14:19.167 : zCorr:0.422 atX:115.30 atY:161.69

23:14:24.103 : N23 G1 E1*77
23:14:24.108 : zCorr:0.402 atX:115.30 atY:161.69
```

And here a full sequence:
```
23:08:12.573 : N40 M82*45
23:08:12.578 : ok 40
23:08:12.578 : Echo:N40 M82

23:08:19.088 : N41 G92 E0*114
23:08:19.091 : ok 41
23:08:19.091 : Echo:N41 G92  E0.0000

23:08:24.995 : N42 G1 X115.296 Y161.692 Z1 F1800*56
23:08:24.998 : ok 42
23:08:25.000 : Split line len:198.58 segments:20
23:08:25.001 : zCorr:0.003 atX:5.77 atY:8.08
23:08:25.004 : zCorr:0.007 atX:11.53 atY:16.16
23:08:25.009 : zCorr:0.020 atX:17.30 atY:24.25
23:08:25.013 : zCorr:0.035 atX:23.06 atY:32.33
23:08:25.017 : zCorr:0.057 atX:28.83 atY:40.42
23:08:25.021 : zCorr:0.085 atX:34.59 atY:48.50
23:08:25.025 : zCorr:0.110 atX:40.36 atY:56.59
23:08:25.030 : zCorr:0.132 atX:46.12 atY:64.67
23:08:25.034 : zCorr:0.160 atX:51.89 atY:72.76
23:08:25.039 : zCorr:0.195 atX:57.65 atY:80.84
23:08:25.044 : zCorr:0.232 atX:63.41 atY:88.92
23:08:25.049 : zCorr:0.265 atX:69.18 atY:97.01
23:08:25.055 : zCorr:0.297 atX:74.94 atY:105.09
23:08:25.059 : zCorr:0.332 atX:80.71 atY:113.18
23:08:25.065 : zCorr:0.365 atX:86.47 atY:121.26
23:08:25.069 : zCorr:0.387 atX:92.24 atY:129.35
23:08:25.074 : zCorr:0.400 atX:98.00 atY:137.43
23:08:25.374 : zCorr:0.410 atX:103.77 atY:145.52
23:08:25.705 : zCorr:0.415 atX:109.53 atY:153.60
23:08:26.037 : zCorr:0.422 atX:115.30 atY:161.69
23:08:26.699 : Echo:N42 G1  X115.30 Y161.69 Z1.00 F1800.00

23:08:32.992 : N43 G1 X115.296 Y161.692 E1 F1800*38
23:08:32.996 : ok 43
23:08:32.998 : zCorr:0.422 atX:115.30 atY:161.69
23:08:33.002 : Echo:N43 G1  X115.30 Y161.69 E1.0000 F1800.00

23:08:39.558 : N44 G1 E-5 F4800*15
23:08:39.561 : ok 44
23:08:39.563 : zCorr:0.402 atX:115.30 atY:161.69
23:08:39.565 : Echo:N44 G1  E-5.0000 F4800.00

23:08:44.658 : N45 G1 X114.082 Y161.135 F10800*72
23:08:44.662 : ok 45
23:08:44.663 : zCorr:0.422 atX:114.08 atY:161.13
23:08:44.667 : Echo:N45 G1  X114.08 Y161.13 F10800.00

23:08:49.677 : N46 G1 E1 F4800*36
23:08:49.680 : ok 46
23:08:49.682 : zCorr:0.402 atX:114.08 atY:161.13
23:08:49.685 : Echo:N46 G1  E1.0000 F4800.00

23:08:54.890 : N47 G1 X112.736 Y160.908 E2 F1800*36
23:08:54.894 : ok 47
23:08:54.896 : zCorr:0.425 atX:112.74 atY:160.91
23:08:54.900 : Echo:N47 G1  X112.74 Y160.91 E2.0000 F1800.00
```

HenryLoenwind, 2017-04-21:
I think I see what's happening here:

If there's no movement in the G1 command, then `Printer::destinationSteps` does not get changed, it keeps its current position. Then the distortion correction is run, which expects `Printer::destinationSteps[Z_AXIS]` to have uncorrected data from the G1 command and corrects it again.

repetier, 2017-04-22:
Yes, I'm pretty sure you are right. It adds offset twice. So result is ok since it goes back but it takes time and is not nice. Have uploaded a fix that should work. Please report if it is working. 

I just remove the optimization for no xyz move if distortion correction is available, so destination now gets reset every time.

HenryLoenwind, 2017-04-23:
The result is ok on paper, but in reality it gives a big blob on every hop...

I'll test it. Thx.

HenryLoenwind, 2017-04-23:
```
07:50:05.373 : Autoretract:0
07:50:09.027 : N19 G1 X115.296 Y161.692 Z1*89
07:50:09.033 : zCorr:0.057 atX:115.30 atY:161.69 atZ:0.92
07:50:35.515 : N20 G1 E-1*99
07:50:35.521 : zCorr:0.057 atX:115.30 atY:161.69 atZ:0.92
07:50:45.385 : N21 G1 X115.296*69
07:50:45.391 : zCorr:0.057 atX:115.30 atY:161.69 atZ:0.92
07:50:56.466 : N22 G1 X115.296*70
07:50:56.471 : zCorr:0.057 atX:115.30 atY:161.69 atZ:0.92
07:51:05.042 : N23 G1 E1*77
07:51:05.047 : zCorr:0.057 atX:115.30 atY:161.69 atZ:0.92
```

looks good

repetier, 2017-04-23:
Great. So I close the issue. If there are more problems feel free to open again.

closed 2017-04-23 by repetier