repetier/RepetierFirmware #670: Z jerk was too low, setting to xxx

clearbucketLabs, 2017-05-05:
This is a bit odd, not sure what triggers this or if it's a bug.    Same goes for other jerk settings.  This is on dev branch.   the expectation is you should be able to set the jerk as low as you want and would accelerate from that point, so even 0 should be possible.   I didn't see this happen on the current stable release so seems new in dev. 

Recv: XY jerk was too low, setting to 10.95
Recv: Z jerk was too low, setting to 1.58

I set XY to 10, and Z to 0.2 as it usually is that low for Z at least and then the firmware rejects it.

repetier, 2017-05-06:
It i s no error, just math. Start speed is 50% of jerk. So jerk must be 2 * minimum speed. Minimum speed depends on steps per mm and acceleration and comes from explicit acceleration integration used. Lower values would result in unstable speed calculations. minimum speed is calculated like this:

float minimumSpeed = accel * sqrt(2.0f / (axisStepsPerMM[X_AXIS]*accel));

closed 2017-05-06 by repetier

clearbucketLabs, 2017-05-23:
Sorry,  why would there be minimum speed?  isn't acceleration and jerk more or less a maximum not a minimum? and it wouldn't speed be determined by the slicer?  why does steps per mm determine speed?  that seems unrelated to how fast you move those steps.  i could move 1 step per year, won't matter if there is 200, 400 or 1000 steps per mm.... ?

repetier, 2017-05-23:
You compute time for next step by explicit integration at current speed. For low speeds the error is so high that it skips several acceleration steps and continues with such a high speed that you will loose steps. It is some math requirement to ensure smooth movement from the start and is a bit difficult to understand if you have not done the math manually in a excel sheet.

macdarren, 2017-05-23:
just trying to tweak some of my settings...can you tell me what the unit for "accel" in the above equation.  I tried mm/sec^2 but that results in a bigger minimum than I would expect.

repetier, 2017-05-23:
float accel = RMath::max(maxAccelerationMMPerSquareSecond[X_AXIS], maxTravelAccelerationMMPerSquareSecond[X_AXIS]);
    float minimumSpeed = accel * sqrt(2.0f / (axisStepsPerMM[X_AXIS]*accel));

so yes it is mm/sec^2

actodefe, 2020-05-10:
> ```
> float accel = RMath::max(maxAccelerationMMPerSquareSecond[X_AXIS], maxTravelAccelerationMMPerSquareSecond[X_AXIS]);
> float minimumSpeed = accel * sqrt(2.0f / (axisStepsPerMM[X_AXIS]*accel));
> ```
> 
> so yes it is mm/sec^2

this can be noob question but can you please explain what the meaning for "sqrt" and "2.0f"?

so if the acceleration value is 1500mm/s2 and the travel acceleration is 3000 mm/s2, what should be the minimum speed value?

repetier, 2020-05-11:
sqrt is square root and 2.0f is value 2 as single precision floating point precision.