repetier/RepetierFirmware #677: inconsistent extrusion for different feed rates

clearbucketLabs, 2017-06-05:
Hello!

was having some issues with some prints, they just seemed a bit off.   i figured maybe i am missing steps, clogged extruder etc....   i have a bunch of nozzles so just put a new one to rule that out plus take a look to make sure it's 100% clear.  it is.   checked for missing steps and current settings etc....

when i do a e-steps calibration, it's consistently off depending on the speed and amount.   So if i extrude 50mm at 3mm/s it's bang on the mark or so,  if i extrude same thing but at 1.5mm/s it extrudes 1 mm more or so... doesn't sound like skipping steps in this case.     if i got faster, 3-4mm/s  it extrudes less than it should. 

Now i can do the test 3 times for each speed, and each test is consistent in the amount it extrudes.  What i expect is to extrude the exact amount regardless of speed except in the case i am doing it do fast and the extruder is skipping naturally.

The firmware has extruder advanced on, but the value is set to 0.    anything to explain this behaviour?   would linear advance still affect this even with it set to 0?  this is the dev branch but whatever is used on the configuration tool about a month ago.    

my next steps is to remove it, and reflash or swap firmware and see.

repetier, 2017-06-06:
This is not really a firmware issue and advance would not make it better, maybe even worse.
What you see is a physical problem and a effect that you could call slippage. With increasing speed you need more force to push the same amount of filament in less time. This increases the force you need and as long as the motor can create the force the difference is not that big but gets bigger the close you go to the limit. Putting the filament through your extruder creates small grooves that push forward. With more force it starts making the grooves just a bit longer in extrusion direction, meaning it did not move it the same a mount. In the end the extruder will either block as it has not enough force of slip out of the grooves.

One other reason for increasing force is if you get faster is, that filament gets less time to heat up and colder filament is even harder to extrude. So in both cases a hotter filament would decrease the force and reduce the effect you have. Of course it will also increase oozing when idle.

Would be interesting to see how the figures change with lets say 5°C more temperature. Maybe you can post a table with the results and if we see a pattern there might be a solution by tweaking temperatures depending on average speed. Not that easy if speed changes all the time, but it might help and that would increase quality I guess.

clearbucketLabs, 2017-06-06:
Hey,

Thanks for the reply.   i didn't think advanced being on but set 0 would affect it but you never know.   

However it seems swapping the driver has fixed it so far at least.

closed 2017-06-06 by clearbucketLabs