repetier/RepetierFirmware #681: DEV Compile error  (Arduino Due, Radds)

clearbucketLabs, 2017-06-25:
I have a previous version of dev that works fine (no idea what version since they're not marked, definately months ago),  but the last one gets a compile error in  arduino 1.5, 1.6 and latest (1.8?).   

```
var/folders/3k/ztw4x2f95lscw8g5g798836c0000gn/T/arduino_build_266333/sketch/SDCard.cpp: In member function 'void SDCard::pausePrint(bool)':
SDCard.cpp:163: error: expected primary-expression before ',' token
                             Printer::memoryE - RETRACT_ON_PAUSE,
                                                                ^

```

Could be me but i've updated all the libs to the latest as well and no go.    uncommenting allows it to compile.

repetier, 2017-06-25:
I think you are using your old configuration.h. You have to redownload it through config tool (if created with it) so it adds new variables. In this case
// Retraction for sd pause over lcd
#define RETRACT_ON_PAUSE 2

seems to be missing in configuration.h but there might be more.

clearbucketLabs, 2017-06-25:
ok, so i put 0 for in the config but it didn't write 0, it was blank.

repetier, 2017-06-26:
Just tested and it worked. Might be that you had a older version of the javascript in cache, but it definitely works after clearing cache and reloading.