repetier/RepetierFirmware #682: X does not stop on end stop, depends on other switch states?

clearbucketLabs, 2017-06-25:
This isn't the first time, but usually it's fine and i never see it.  But i swapped to use Z-Max, i was using probe as Z-Min but it's problematic and un-ideal.     Using the Latest DEV, and home XYZ.  Cartesian, Due+Radds.

 Anyway....

 1. I flash firmware
 2.  Click Home XY
 3.  X does not honor the min switch, and stops and homes on it's own without pressing the switch (regardless it ignored it anyway, like a software End Stop which is OFF.  it would drive the extruder into the switch and grind away then stop eventually)
4. Y Homes fine after X thinks it homed.  (i had to remove the belt so it didn't break the printer)
5. Also, in addition Z moves up 1 mm, it shouldn't move if i home XY ONLY (this is part of the issue)
6. If i HOME Z, it triggers Z-MAX and works as expected.
7. After Homing Z, X Min Now Works again. (YAY?)
8.  Hit Reset on the controller (Due/Radds).
9.  Click Home XY again,  X Switch Works but Z Moves up 1 mm again.
10.  Now X min no longer functions until i re-home Z.

This is weird... IMO i don't think ANY setting should cause this behaviour regardless so i think it's a Bug.  it could be triggered by firmware settings but there should be preventative measures here.  

Here is my config though.... Hopefully it is config related, all i've changed since before was a Z-Max, home XYZ and Z-Min = false and Z-Min Is PROBE (software z-max is off, etc... the usual flags).   Prior my previous settings worked fine, except in the case where if i moved the extruder manually and tried to home it later, it would move 10 mm opposite direction until i move it back to trigger the switch.  a bug no doubt.

or eeprom, didn't think of that but still. 

Relevant settings:

```
#define DRIVE_SYSTEM 0
#define XAXIS_STEPS_PER_MM 80
#define YAXIS_STEPS_PER_MM 80
#define ZAXIS_STEPS_PER_MM 400
#define EXTRUDER_FAN_COOL_TEMP 50
#define PDM_FOR_EXTRUDER 0
#define PDM_FOR_COOLER 0
#define DECOUPLING_TEST_MAX_HOLD_VARIANCE 5
#define DECOUPLING_TEST_MIN_TEMP_RISE 1
#define KILL_IF_SENSOR_DEFECT 0
#define RETRACT_ON_PAUSE 0
#define PAUSE_START_COMMANDS ""
#define PAUSE_END_COMMANDS ""
#define SHARED_EXTRUDER_HEATER 0
#define EXT0_X_OFFSET 0
#define EXT0_Y_OFFSET 0
#define EXT0_Z_OFFSET 0
#define EXT0_STEPS_PER_MM 418.5
#define EXT0_TEMPSENSOR_TYPE 8
#define EXT0_TEMPSENSOR_PIN TEMP_0_PIN
#define EXT0_HEATER_PIN HEATER_0_PIN
#define EXT0_STEP_PIN ORIG_E0_STEP_PIN
#define EXT0_DIR_PIN ORIG_E0_DIR_PIN
#define EXT0_INVERSE 0
#define EXT0_ENABLE_PIN ORIG_E0_ENABLE_PIN
#define EXT0_ENABLE_ON 0
#define EXT0_MIRROR_STEPPER 0
#define EXT0_STEP2_PIN ORIG_E0_STEP_PIN
#define EXT0_DIR2_PIN ORIG_E0_DIR_PIN
#define EXT0_INVERSE2 0
#define EXT0_ENABLE2_PIN ORIG_E0_ENABLE_PIN
#define EXT0_MAX_FEEDRATE 100
#define EXT0_MAX_START_FEEDRATE 20
#define EXT0_MAX_ACCELERATION 5000
#define EXT0_HEAT_MANAGER 3
#define EXT0_PREHEAT_TEMP 190
#define EXT0_WATCHPERIOD 1
#define EXT0_PID_INTEGRAL_DRIVE_MAX 230
#define EXT0_PID_INTEGRAL_DRIVE_MIN 40
#define EXT0_PID_PGAIN_OR_DEAD_TIME 7
#define EXT0_PID_I 2
#define EXT0_PID_D 40
#define EXT0_PID_MAX 255
#define EXT0_ADVANCE_K 0
#define EXT0_ADVANCE_L 0
#define EXT0_ADVANCE_BACKLASH_STEPS 0
[... 247 lines cut ...]
#define Z_PROBE_Y1 20
#define Z_PROBE_X2 160
#define Z_PROBE_Y2 20
#define Z_PROBE_X3 100
#define Z_PROBE_Y3 160
#define BED_LEVELING_METHOD 1
#define BED_CORRECTION_METHOD 0
#define BED_LEVELING_GRID_SIZE 5
#define BED_LEVELING_REPETITIONS 5
#define BED_MOTOR_1_X 0
#define BED_MOTOR_1_Y 0
#define BED_MOTOR_2_X 200
#define BED_MOTOR_2_Y 0
#define BED_MOTOR_3_X 100
#define BED_MOTOR_3_Y 200
#define BENDING_CORRECTION_A 0
#define BENDING_CORRECTION_B 0
#define BENDING_CORRECTION_C 0
#define FEATURE_AXISCOMP 1
#define AXISCOMP_TANXY 0
#define AXISCOMP_TANYZ 0
#define AXISCOMP_TANXZ 0

```
`

clearbucketLabs, 2017-06-25:
note that if i remove back on end stop, and keep Z triggered X min stays working after reset.  if it moves back un-triggering the Z-Max or a move it manually and reset controller, then it ignores X-Min until i home Z first.

repetier, 2017-06-26:
In first instance is sounds like a hardware problem. If x min working depends on z min being triggered then something is wrong. It should be an independent pin. M119 should show you if it is triggered if z min is not.

The z move after xy homing is another problem. For z min there are many ways to get extra moves, for z max it might be the autoleveling getting you back to starting z position. If you have 2 homes in a row at same position the z move should then vanish on second run.

clearbucketLabs, 2017-06-26:
I can't say what HW problem would do it in this case, i've swapped the due board, erased everything and re-flashed, same thing.    

M119 reports as it should.  I also hooked up the scope, looks normal.  I removed the switch to reduce possible crosstalk issue. 

it also works fine in the other case of Z getting triggered First.  it does not have to STAY triggered,  it homes Z then moves Down 5mm,  it just has to home Z First instead, THEN it works fine every time.  

if it's remembering State, and affected by a reset too it's difficult to say it's hardware without debugging it to see what the MCU sees.  especially if it works if i re-flash with my previous configuration of XYZT and Z-min as probe.

clearbucketLabs, 2017-09-11:
Revisting this... definately some odd bug.   i've had it happen (only with X)  even with XYZT.  if i home Z, it homes X/Y fine then moves up 10mm, probes Z-Min and goes back into the home position.   If i disabled motors for example and manually move X to check bed and nozzle height etc... then home only X/Y (G28 X Y)  then the X keeps moving and rams into the switch, no matter how many times i run home X/Y.    BUT if i do home "Z"  which auto-matically homes XY First, it works perfectly fine....  

I've also tried other firmware and hitting end stops is a non-issue.  everything is fine hardware wise.   it behaves like a sw bug so i assume that's what it is.   note i've never saw this with a Z-min endstop...  maybe it's something to do with the Probe and it's only the x axis....