repetier/RepetierFirmware #683: cannot get a decent upload to my board

neovin87, 2017-06-28:
hi, every time i upload dev, .92 or master i get this error. I have managed to keep repeating uploads untill it works but seems fewer and far between calibration is impossible.

Sketch uses 120,266 bytes (47%) of program storage space. Maximum is 253,952 bytes.
Global variables use 6,136 bytes (74%) of dynamic memory, leaving 2,056 bytes for local variables. Maximum is 8,192 bytes.
avrdude: verification error, first mismatch at byte 0x023b
         0x5f != 0x4f
avrdude: verification error; content mismatch
avrdude: verification error; content mismatch

is there a fix and would this be causing printer breaking bugs?

Thanks, Neo.

repetier, 2017-06-29:
Recently I had a user with same problem. Result was he had a board that could cope with 256KB files but the bootloader did not support this. After replacing bootloader (not so easy) he could upload again. Here you are below the critical length of 122880 byte so it might be something different.

If the verification error always at a different address? Generally spoken a verification error says you uploaded something and I stored something different. If that byte is just part of a string nothing bad will happen if you run. If it is part of assembler code and you run that function where it is it will most probably crash or do something unexpected. This is no firmware problem but a problem that flashing does not work on board. If address changes it is just communication error probably and retrying could help.

neovin87, 2017-06-30:
Tried many times and had it work some and not.
Any ideas what I could try, I was getting amazing prints out of my machine until I rebuilt and needed to change settings

neovin87, 2017-06-30:
And yes it is always at different byte locations

repetier, 2017-07-01:
If possible do not enable main power so it is only powered over usb. Sometimes a different, shorter or better shielded usb cable makes a difference. After all it is a big program so if communication is not that stable it is likely to get an error. During print these are normally filtered by checksums.

neovin87, 2017-07-13:
Well I found the cause of this agony.
I repeated testing, trying diffrent cables ports and configuration to no avail.
After repeatedly trying i replaced the mega2650 with a new on and that sorted it all out.
I will test the board and try and locate if the issue was the 16u4 or the 2650 but hope this infomation helps someone else struggling through this.
Thanks for your help, Gavin

closed 2017-07-13 by neovin87