repetier/RepetierFirmware #685: Three Z-Endstops possible?

siggi60, 2017-07-12:
Hi everybody,
i'm refeering to this [Issue](https://github.com/repetier/Repetier-Firmware/issues/528)
ist it also easy to setup/integrate a third Z-Endstop? I don't really believe that all is done by adding some lines like them:

#define MINMAX_HARDWARE_ENDSTOP_Z3 1
#define Z3_MINMAX_PIN 16
#define ENDSTOP_Z3_MINMAX_INVERTING 0
#define ENDSTOP_PULLUP_Z3_MINMAX 0

My intention is it might be better/more precise to use three optical or hall endstops then one inductive sensor to do mabl.

Thanks for reading

repetier, 2017-07-12:
No it is not that easy. And it is not supported as far as I remember. We have adjusted z homing to use dual endstop homing.  You would need to rewrite the endstop handling and homing to add a third endstop and then you will see that it is hard to level with endstops. Better is to use a z probe instead with mechanical bed leveling since you need 3individual controlled z motors anyway.

siggi60, 2017-07-12:
Thanks for that quick response.
I have three independed Z-Motors so i know MABL.
My thinking was the following as example:
1) Z goes up with tilted bed to do mabl - Z3 has first contact - everything stopps
2) M3 goes a bit down - mabl continues
3) now Z2 has contact - everything stopps
4) M2 goes a bit down - mabl continues and repeat that until Z1 has first contact
5) mabl can be done right as intended

Probes are sometimes imprecisely, so i was hoping my idea was easy to implement.
A pitty that's not. I should go and learn C++ and then.... *gg*

Wait, another idea :)
How about one endstop and two "flags" formerly known as Z2 and Z3?
Z2 is triggered? OK, M2 go down.

So endstop handling can be as it is and only needs to read the state of other pins set by user.
Maybe the better way?