repetier/RepetierFirmware #686: Pause before probing

DjDemonD, 2017-07-13:
Hi all,

I now market and sell an open-source piezo Z-probe. More details can be found online at reprap forum, just see the general forum main page, I won't leave any spam here.

What we are interested in is having a configurable pause before a probing dive takes place. RRF already has this as DC42 and T3P3 are making a nozzle contact probe based on piezo like we do, and the M558 command now has an R (recovery) parameter which is pause (in seconds) after positioning over a probing point before diving to probe. A value of 0.2sec is usually enough. It allows mechanical noise to settle before probing an reduces/eliminates false triggers from this very sensitive probe.

These things are becoming quite popular now as they are by far the most convenient and accurate z-probe available.

Would you consider implementing it for the benefit of the now hundreds of people using piezo z probes? I am not certain if your firmware already has such a switch that can be used or if the code already builds in a recovery before probing, but would be very grateful if you would consider it.

Kind regards
DjDemonD/Simon
Precisionpiezo.co.uk

repetier, 2017-07-13:
In dev version I have added
Z_PROBE_DELAY
That is the delay in ms before going down. I guess that is exactly what you need. Please test if it works for you or needs any changes.

DjDemonD, 2017-07-13:
That sounds exactly what we need, so many thanks.

I have not got a machine with your firmware on it myself, but I will post to see if some of the other piezo users have and can verify it has the functionality we are requesting.

closed 2017-07-13 by DjDemonD

repetier, 2017-07-13:
Great. Let me know if something needs to be added.

pumpal, 2017-11-02:
I am on this at the moment. Installing a PrecisionPiezo probe on my HPCE coreXY printer (RADDS+DUE) . 
I downloaded the current dev and ported the change back to 0.92.9. 
I think in BedLeveling.cpp the Z_PROBE_DELAY section must be moved ABOVE the following block:    <waitForZProbeStart();
    Endstops::update();
    Endstops::update();
    if(Endstops::zProbe()) {
        Com::printErrorFLN(PSTR("z-probe triggered before starting probing."));
        return ILLEGAL_Z_PROBE;
    }>
The idea of the whole delay is to give time to the Z prove to rest after a rapid XY move. It is very sensitive and the XY travel often triggers it On. 
In my case that made the autoleveling succeed, where it was constantly failing with "z-probe triggered before starting probing" before adding the Z_PROBE_DELAY code. 

Other then that, seems that the probe works charms with Repetier (@0.92.9 atm)  - both as Z min for homing and for autoleveling. 

Thanks both to the PrecisionPiezo team for the great device/design and to Repetier for the great firmware, which I adore!

repetier, 2017-11-02:
Are you sure?

```
#if defined(Z_PROBE_DELAY) && Z_PROBE_DELAY > 0
        HAL::delayMilliseconds(Z_PROBE_DELAY);
#endif
        PrintLine::moveRelativeDistanceInSteps(0, 0, -probeDepth, 0, EEPROM::zProbeSpeed(), true, true);
```
As you see it just before it starts the move down and inside the repeat loop.

The position you say is what z probe gets called. At that point is should already be untriggered for a second from the move to probing xy position. Also it is outside the repetitions loop so for repetitions it would not wait.

Please try the original version if it works. Your version would work with only one measurement while mine should work for both 1 and more repetitions.

pumpal, 2017-11-02:
Well, I  was getting this error:
Com::printErrorFLN(PSTR("z-probe triggered before starting probing."));
The delay must be added just after the XY move ends. 
Moving the delay section before the section where it checks the probe solved it for me. 
But you may be right - this is only my case where the Z moves independently from  XY. 
Maybe it must not be moved but remain in the loop and added once again before the probe checks ..

repetier, 2017-11-02:
Ok, also I don't know how it comes so fast there that it is still triggered I have added it in addition to also be save on the start condition. Will be part of next update.

pumpal, 2017-11-02:
This particular probe is activated by the hotend move (the piezo produces signal when it bends). So because of the mass/inertia I think there is a short false triggering caused by the XY move stop - it is not "still triggered" but instead it was just triggered after the move ends.

pumpal, 2017-12-18:
Hi again. 
I am successfully using this probe with Repetier 1dev_11/21 with very good deviation (don't have the exact numbers by hand at the moment). 
I have a question though - in order to get the right height when printing, I have to set either bed coating or z-babystepping to -0.08 (minus 0.08).  What does that mean ?  Is it Normal/Ok or is an indication of some mechanical problem with the probe/printer ? 
Thanks in advance.

repetier, 2017-12-18:
That means you z probe height is off by 0.08mm and you selected a different way to compensate the error.

DjDemonD, 2017-12-18:
Okay the piezo system has to have a small degree of compliance to trigger, the piezo disc bend/compresses on nozzle contact. You are measuring this compliance. Our online docs state that a z offset of -0.1 is normal and expected. Simon (Precision Piezo).

pumpal, 2017-12-18:
Simon, I though you said a +0.1 is normal and expected. That would mean it triggers a little beneath the bed, right? In my case I have to set a negative value...

repetier, 2017-12-18:
No, negative means it triggers after touching bed. So piezo, force feedback, switch behind extruder all have negative z probe height, also quite small normally.

DjDemonD, 2017-12-18:
I am talking about the offset I use in RRF firmware G31 command which needs to negative for a probe that triggers below the bed, and positive for a probe which triggers above the bed (like IR, inductive). In slic3r I'd use a positive offset instead. SO suffice it to say the offset depending on where you are entering it is usually between 0.1-0.15mm depending on what piezo sensor configuration you have, and is usually in my experience negative.

OneDayFlie, 2018-11-12:
Is there any info how to setup the piezo? Can't find a good manual...  So if anyone can help me? Thanks!

DjDemonD, 2018-11-12:
https://www.precisionpiezo.co.uk/documents
Not particularly repetier-specific but then it's a fixed z probe can be setup depending on the piezo board revision as NC or NO.

pumpal, 2018-11-13:
Here is my Configuration.h from 1.2 
[Configuration.zip](https://github.com/repetier/Repetier-Firmware/files/2574127/Configuration.zip)

if it can be of any help.

OneDayFlie, 2018-11-13:
Thanks.
Yeah somehow i got it working without changing settings.... The semi automatic bed leveling settings that are allready on it are working with it.
Only need to change te speeds, because it is ginving me a false trigger ;)