repetier/RepetierFirmware #687: PID controller :: I component of the PID controller

Nibbels, 2017-07-18:
_One word to my limits: I only work with an old version of repetier.
It is the Conrad Renkforce RFx000-Firmware. But I guess this is something that you might be interested in._

I wondered why the PID did not work properly.
The printer tends to have an persistent overshoot. That let me thinking. I learned that the I-Component of the PID should eat up any possible leftover.
Later I found this thread http://forum.repetier.com/discussion/3188/possible-bug-in-pid-routine too, while researching the problem.

I began compare to update the PID routines and experimented with the autotune function. But even if I used the conservative KP-KI-KD-Ruleset ("No Overshoot" within the comment), I just failed. The printer tended to never reach the temperature or to stay atop. When I tuned the values manually KI kind of did nothing. I tuned it to KI=100 and in some setting that seemd to be the same result as KI=1.

I know about "PID drive max" and "PID drive min".
I know about "PID max value"

Long story short: Could you maybe check some detail about constraining the I component?
https://github.com/repetier/Repetier-Firmware/blob/development/src/ArduinoAVR/Repetier/Extruder.cpp#L674
When the I-component is always positive it cannot eat away a positive temperature offset.
I did patch this to 
`tempIStateLimitMin = (float)pidDriveMin * -10.0f / pidIGain;`
Patched, the limited range of the Integral sum is from **"-Min" to "+Max"** and therefore can control the temperature better.

My own tests were quite promising. Can anybode confirm this for me?

Greetings from Stuttgart / Germany

Nibbels, 2017-07-19:
Hint: According to this table:
https://en.wikipedia.org/wiki/Ziegler%E2%80%93Nichols_method @ "some overshoot"
The following https://github.com/repetier/Repetier-Firmware/blob/development/src/ArduinoAVR/Repetier/Extruder.cpp#L2541 (commented out but there) present for tuning the PID is missing the factor 2.

I guess it should be Ki = 2*Kp/Tu.

some overshoot | {0.33K_{u}} | {T_{u}/2} | {T_{u}/3}
-- | -- | -- | --

I did not open another issue because this is a quite minor little flaw.

Greetings

repetier, 2017-07-19:
The function was copied from Marlin or sprinter I guess, so nothing I have good knowledge with. Just compared with marlin and they have the same error. Comparing with wikipedia I see that they even write I = Tu/2 and we compute Tu * 2 while computation for D is the same, so I assume Tu has the correct meaning used in the wikipedia article, also they do not show how to get Tu and Ku. So I'm now wondering if they were always wrong or if this is a change to make them work better with extruders.

Regarding tempIStateLimitMin - well that is something I added as I saw that during first raise the I term could accumulate to much negative value and I assumed it should always such that the integral part alone leads to a range between drive min and max. Simply because if your are exactly on the target line P is 0 and damping is zero. So I is the PWM we really need and P and D are just helpers to get there fast in waves. 
So if you say allowing negative values makes it control better, the question is if we should contains it at all.

So interesting thing would be to test the 3 approximation methods with and without constrain and like wikipedia says or current original and decide what is best.

For that I will also add a C parameter to select classic, some and no overshoot. Will be interesting to see what will happen.

Nibbels, 2017-07-19:
My guess is that the meaning of Ku and Tu are the following:
there (https://en.wikipedia.org/wiki/Ziegler%E2%80%93Nichols_method) is this formula:

u(t) = KP * ( ... + 1/Ti _Integral_ + TD _Differential_ )

And I understand it so that 
KP is there
KP / Ti is something and 
KP * TD is something

with: **Ti = Tu/2**
KP / Tu * 2

with **TD = Tu / 8**
KP * Tu / 8

Thathow we get something like
```
Kp = 0.6f*Ku;          //0.6 KRkrit  
Ki = 2.0f*Kp/Tu;       //0.5 Tkrit  
Kd = Kp*Tu/8.0f;       //0.125 Tkrit
```
within the code.
But careful I am no expert. I just did sort some structure to get this idea. :)

About the testing: I had the same idea and just added 6 additional rules using `hasJ()` within my branch. 6 = P-control
5 = PI-control
4 = PD-control
3 = PID-control "no overshoot"
2 = PID-control "some overshoot"
1 = PID-control "Pessen Integral Rule"
0 = PID-control "Classic Ziegler-Nichols"

You can see it here near the places of this last commit:
https://github.com/Nibbels/Repetier-Firmware/commit/bd8d509f1ac2c8b6884d265df4d950991019a07a

repetier, 2017-07-19:
I will do some more tests tomorrow. Currentyl removing the i constrain causes big problems as i is 0 so it drops too fast since it could not adjust during full temp heating. Now I preset it to the max value and that creates the massive overshoot I read about triggering decouple tests :-(. So will see where I end up so I do not upset too many users with new behaviour. Even if it is then better the will complain since their values were adjusted to old behaviour so may be worse then.

Nibbels, 2017-07-19:
I totally understand.
Maybe some control engineer sees this in future and knows exactly what to do. 

For me a new autotune with _Pessen Integral Rule_ works quite well now.
(I put Drive Max to 100 and Drive Min to (-)5 and Min without 10.0f scaling.)
I have exact temperature readings now. The temperatures go to 230.0°C as example, not 231.7°C till 232,8°C anymore.

There is a overshoot rightnow.
There is a undershoot rightnow. 
But the constant overshoot is gone.

Greetings

Nibbels, 2017-07-19:
[image]
[image]
[image]
[image]
[image]

repetier, 2017-07-23:
I found a good documentation of PID here:
http://m.eet.com/media/1112634/f-wescot.pdf

it also has the min/max I use included. Based on this I have improved the handling a bit but min/max stay as they are. But since pidDriveMax/min are only used there you can simply set these to any value you like without modifying the code. But my many tests showed that it is good to have these.

I now update temperature deviation every second so it gets more stable and as a benefit we now require less memory:-)

closed 2017-07-23 by repetier

Nibbels, 2017-07-23:
Ok :)

For future lookups the essence: 

A PID is a P+I+D control. 
If I cannot be negative, it is some type of "P+D"+"someI"-control.
The stability depends on the very good choice of PD-Control parameters where I does not really matter because it is 0.

My friend formerly had problems with 60Watts within the very small Ultimaker-Heaterblock. The problems are gone with this patch - mine too. I maybe will implement another EEPROM-Value which can be set to -1 to correct the direction of PID-Drive-Min. That will make real PID-controll possible but most of the People will not notice it.

Nevertheless I see your point here! Thank you for your review of the problem.

LG

Nibbels, 2017-07-23:
About your patch: https://github.com/repetier/Repetier-Firmware/commit/6bdd97e83b656d15cafe51958ba1504e1bfb97de

Resulting use is "act->lastTemperatureC - act->temperatureC". This is the temperature we are using within PID-Control. This is updated only once a second, If I read this right.
If thats true, the D-Part of P-I-D thinks that the differential temperature is constant 1 second long each second?
(And the age of this values will be one utill two seconds at runtime. So there is a delay of 1 second minimum.)

If I am right a huge change of temperature (wrong reading etc.) at updatetime will steer the deviation-part of the control-calculation to a wrong temperature one second long.

Shouldnt there be at least one currentTemperature involved at "manageTemperature"-Time?
This might be totally fine if you can compete with those readings at runtime :)
[image]
I posted this @ http://www.rf1000.de/viewtopic.php?f=25&t=1868&p=19186#p19186

Greetings

repetier, 2017-07-23:
Yes, the D part is the biggest change. Previously it used difference between 300ms timing. This also means the PID is updated quite often. If you read the paper you will see that too much updates are a bad idea. So this goes down to 1s update for D and I will only update 10% so it sums up to 100% in one second and P is not time critical anyhow, so updates every 100ms.
The other improvement is that the 100ms temperatures are preserved until being processed so ADC will not overwrite them if it took a bit longer to run the temperature controller.

As I said it is fine to have pidDriveMin set to -1 if you need your negative value. The constrain is just to make I not flow out of range. Max. we can have is 255 so having I at 1000 just means it takes a very long time to get it < 255.

Temperature curve looks quite good also the output has some big jumps. On my test extruder it was much smoother. But I'm glad it now works where it did not before:-)

Nibbels, 2017-07-23:
Oh yeah...
less dT/dt and less turbulence in steering.
[image]
I just changed the tempArray to wasteful 16 floats and capped at act->tempPointer &= 15; and dropped the reciproke to 1/1500ms.
It cannot be better now. I did not really get this until you told me.

Thanks alot!