repetier/RepetierFirmware #691: Possible Overflow

Nibbels, 2017-07-30:
Hello :)
(Again one word to my limits: I only work with an old version of repetier.
It is the Conrad Renkforce RFx000-Firmware. I am not from Conrad, I just mod the Firmware to my needs and bugfix it.)

There is some possible flaw which might be some Problem in combination with others.
https://github.com/repetier/Repetier-Firmware/blob/master/src/ArduinoAVR/Repetier/HAL.cpp#L1030
https://github.com/repetier/Repetier-Firmware/blob/development/src/ArduinoAVR/Repetier/HAL.cpp
`if(++osAnalogInputPos >= ANALOG_INPUTS) osAnalogInputPos = 0;`

Some Lines atop is
https://github.com/repetier/Repetier-Firmware/blob/development/src/ArduinoAVR/Repetier/HAL.cpp#L1013
`HAL::allowInterrupts();`

I guess you already know what I am referring to.
My chip is some Arduino 2560. When I changed details and some frequency condition met the interrupts I definitly had an overflow in my firmware. osAnalogInputPos reached the number ANALOG_INPUTS. I saw this with my own eyes because I did something like "if(var > max)  max = var; -> output max later"
When I counted some "countstart++" at the beginning of the Interrupt-Routine In the end it had much higher values than "countend++" at the end of the same Interrupt-Routine.
What I want to say is, that in my special case the Routine sometimes might be startet and does not really end.
I found 3 independend fixes to my overflow problem.
https://github.com/RF1000community/Repetier-Firmware/commit/3eddde7b8029ef597dcc9e18cfb895a5956ef9c6
- Doorkeeper for PWM_TIMER_VECTOR
- forbid-allow-Interrupt for Temperature-Buildup
- Never increment osAnalogInputPos atop ANALOG_INPUTS not even 1ys long.

At this point I am very very interested in any piece of Information what might be connected to my bug and if you fixed something related to this in earlier releases since 2014.
_My problems might also related to other problems that come by having this old version of the firmware modded._

Greetings from Stuttgart

Nibbels, 2017-07-30:
Thanks for your second opinion on that!
I move my code to your solution.

Your explainations helped alot to be more confident with those specific interrupt thoughts.
This fix did totally eliminate temperature spikes on my side.

:)

Nibbels, 2017-08-06:
I have been reading a bit further:
This comment is not because I have or see any more problems, I just thought that using nextPlannerIndex(..) within the following functions would be much nicer:

https://github.com/repetier/Repetier-Firmware/blob/development/src/ArduinoAVR/Repetier/motion.h#L622
```
    static INLINE void previousPlannerIndex(ufast8_t &p) {
        p = (p ? p - 1 : PRINTLINE_CACHE_SIZE - 1);
    }
    static INLINE void nextPlannerIndex(ufast8_t& p) {
        p = (p == PRINTLINE_CACHE_SIZE - 1 ? 0 : p + 1);
    }
```
Is there a specific reason why **nextPlannerIndex()** is not being used here?
https://github.com/repetier/Repetier-Firmware/blob/development/src/ArduinoAVR/Repetier/motion.h#L587
```
    static INLINE void pushLine() {
        linesWritePos++;
        if(linesWritePos >= PRINTLINE_CACHE_SIZE) linesWritePos = 0;
        Printer::setMenuMode(MENU_MODE_PRINTING, true);
        InterruptProtectedBlock noInts;
        linesCount++;
    }

```
and here?
https://github.com/repetier/Repetier-Firmware/blob/development/src/ArduinoAVR/Repetier/motion.h#L575
```
    static INLINE void removeCurrentLineForbidInterrupt() {
        linesPos++;
        if(linesPos >= PRINTLINE_CACHE_SIZE) linesPos = 0;
        cur = NULL;
#if CPU_ARCH == ARCH_ARM
        nlFlag = false;
#endif
        HAL::forbidInterrupts();
        --linesCount;
        if(!linesCount)
            Printer::setMenuMode(MENU_MODE_PRINTING, Printer::isPrinting());
    }
```
I guess I can exchange those linesPos++; and IF-Line safely with the nextPlannerIndex(linesPos)?

Greetings

repetier, 2017-08-06:
If you change
 static INLINE void nextPlannerIndex(ufast8_t& p) {
        p = (p == PRINTLINE_CACHE_SIZE - 1 ? 0 : p + 1);
    }
to
 static INLINE void nextPlannerIndex(ufast8_t& p) {
        p = (p >= PRINTLINE_CACHE_SIZE - 1 ? 0 : p + 1);
    }

it would be the same, also the > case should never really happen or we are in trouble. Will also change it for better readability.

Nibbels, 2017-08-06:
**>=** .. nice!

Thanks!