repetier/RepetierFirmware #693: Slight error in code and SPI improvement

stecor7001, 2017-08-02:
Hello Both,

First of all thank you for a great piece of software! I wanted to express my thanks for a (in my opinion) solid and thought out implementation. I am trying to use Repetier FW for a DUE + modified RAMPS 1.4 setup. During this, I came across the following item:

Branch: DEV
Time: now (Aug 2nd)
Issue: Slight code typo
Impact: Compile fails
File: ui.cpp
Line: 2809
Issue description: Original line reads:
PrintLine::moveRelativeUI_ACTION_Z_OFFSETDistanceInStepsReal(0, steps, 0, 0, Printer::maxFeedrate[Y_AXIS], false, false);

This fails compile with the message:
ui.cpp:2809: error: 'moveRelativeUI_ACTION_Z_OFFSETDistanceInStepsReal' is not a member of 'PrintLine'

Issue seems to be due to a stray **UI_ACTION_Z_OFFSET** within the function name; It should read:
PrintLine::moveRelativeDistanceInStepsReal(0, steps, 0, 0, Printer::maxFeedrate[Y_AXIS], false, false);

---------------------------------------------------------------

Proposed improvement:
--------------------------

When using the define **NONSTANDARD_SDSS** config value, in HAL.cpp, the associated HW SDSS pin is still initialized as output in file HALL.cpp line 385.

Proposed solution:
Do not initialize CS pin - alter the relevant section (line 385) including a compile conditional to ensure that HW SDSS is only setup in standard cases (where the nonstandard directive is not issued), e.g like below:
#ifndef NONSTANDARD_SDSS
    PIO_Configure(
      g_APinDescription[SPI_PIN].pPort,
      g_APinDescription[SPI_PIN].ulPinType,
      g_APinDescription[SPI_PIN].ulPin,
      g_APinDescription[SPI_PIN].ulPinConfiguration);
#endif

Tested, seems to work OK.

Additionally configure SPI so that the CS line is controlled in SW, however I cannot seem to find this option - which is available for STM32 but seems not to be possible for SAM3X.

Thank you!

repetier, 2017-08-02:
The error in ui.cpp is clear and will be fixed in next update.

The NONSTANDARD_SDSS is not so clear. Isn't the SPI_PIN always a output pin? I mean it is the slave select so initializing it as output does not look wrong. When is it wrong to make it an output pin?

I also never found an option to not use HW SS, so I thought it was a requirement for the due.

stecor7001, 2017-08-02:
That was quick :)

Re: SPI - on the due, the SPI CS (SDSS) defaults to D10 for SPI0. There are times when this pin may be taken - e.g. the RAMPS board where D10 controls Heater0.  The issue here is not that the SPI CS pin *must* be an output, which it obviously must be. It is that under the current config, the HW SPI CS pin is always activated. It is usually possible to switch between a HW-controlled CS pin and a SW-controlled CS pin for SPI. 

SDFAT folks already provided SW-control via the NONSTANDARD_SDSS directive, which, if implemented should forego the HW CS pin init. Obviously, Please see below an excerpt from the SdFat.cpp implementation for spiReceive, implementing either HW or SW control for CS:

/** SPI receive a byte */
static uint8_t spiRec() {
#ifdef NONSTANDARD_SDSS
    WRITE(SDSS, LOW);
    uint8_t retval = HAL::spiReceive();
    WRITE(SDSS, HIGH);
    return retval;
#else	
    return HAL::spiReceive();
#endif	

 Does this make sense?

repetier, 2017-08-02:
Actually it was me having added the NONSTANDARD_SDSS to SdFat so I do not count on it being initialized:-)

Having D10 as standard for RAMPS is of course a bad thing and I understand that this should be avoided. Just not so convinced that your solution has side effects in that case.

Which board are you using? 408?
I see HW SPI is 10 software 53. What I do not see is setting
SPI_PIN to a pin number. So I wonder just what is happening then resp. what number it was set to. My guess is that at the end of pins.h it should be set to SDSS when it select NONSTANDRAD_SDSS. That way it gets defined for right pin and not D10.

stecor7001, 2017-08-02:
Ah, so you implemented it? Well done then!

You are right of course. Spi_pin is only set when hw spi is set (pins 10, 52,etc). However if using the conditional above, it would then not be needed, correct?

Also, spi_pin is duplicated with sdss. One of them is too much :) Indeed I added the pin init code in hal.cpp (added an else to the ifndef) and it works no issues. 

I am using my own board def, which I built basically using 33 and adding the bits which were missing (adc channels changed to 3&4) and added i2c eeprom. I could not find a matching board, and it was straightforward to create my own userpins.h (btw pins.h is so complex, I like Marlin's one board per file approach better here)

Sorry I have to say it, but Atmel's implementation of spi is... well... unfortunate. Looks like the automatic cs handling costs at least 6 clocks per transfer, which is a bummer.

Also was surprised to see the performance delta hw vs sw spi when dealing with the sd card. Sw spi is so slooooow I could not believe it. Did you look into using hw spi for a 12864 display? It might be a bit of a performance improvement.

Thanks for your time and your replies.