repetier/RepetierFirmware #698: Feature Request/Looking FWD

stecor7001, 2017-08-08:
Hello All,
Running a Due+RAMPS+DRV8825 setup. I hit a wall with the DRV8825 in that the STEP signals do not follow TI's requirement of 2us min value. As such, the Y axis motor was not moving at all, while X was losing 70+% of its steps. Issue is apparent with A4988 drivers as well.

This was patched through the STEPPER_HIGH_DELAY config item (set to 1us), however this causes dead-time (the delay function is blocking). Please consider altering the step logic because as people move away from AVR to faster CPUs, more time will be wasted. Quick example: delay 1us, print speed of 90mm/s, 16x microstepping, 160 steps/mm results in: 90[mm/s]x160[steps/mm]x1[us] = 14.4 [ms] delay time per axis. This is roughly 1.000.000 cpu cycles wasted in delay. If both axes are moving, I think (haven't checked) delays are cumulative.

It should be possible to have two interrupts per step spaced so that there is no requirement to run a delay loop, rather the second interrupt is delayed until the specified amount of time passed.

Also, may be worth looking into why the X steps are slower than Y. Is there a reason why there is asymmetry?

Thank you!

repetier, 2017-08-08:
Delays are not cummulative. We first set all step signals, then the delay comes and then we unset all signals. This also is the reason x has more high time then y and z.

Splitting interrupt makes only sense if the interrupt overhead is lower then the delay. I don' t think that this is the case even for the due.

stecor7001, 2017-08-09:
Okay, thanks for the explanation - the X vs Y delay was puzzling me.

boelle, 2018-01-15:
@repetier would it help you if I read all the open issues and comment on those that can be closed?