repetier/RepetierFirmware #706: 1.0.0dev UI_DYNAMIC_ENCODER_SPEED no effect

PvtPrivate, 2017-10-14:
"Allows faster value selection by turning encoder faster. (UI_DYNAMIC_ENCODER_SPEED)"
that setting doesn't seem to have any effect on the latest dev version.

repetier, 2017-10-14:
It is not used any more for menu action since that was too unprecise. But it should be still in use e.g. for speed multiplier

PvtPrivate, 2017-10-15:
Scrolling through the SD card to the bottom was pretty painful if you have a filled SD card (100+ gcodes)

repetier, 2017-10-15:
But exactly the sd card scrolling was also too fast for many so we needed to take it out here. So the problem is that what one likes do others not like. One group always loses :-(

But you can go to ui.cpp line 2753

```
#if SDSUPPORT
    if(mtype == UI_MENU_TYPE_FILE_SELECTOR) { // SD listing
        if((UI_INVERT_MENU_DIRECTION && next < 0) || (!UI_INVERT_MENU_DIRECTION && next > 0)) {
            menuPos[menuLevel] += 1; // abs(next);
            if(menuPos[menuLevel] > nFilesOnCard) menuPos[menuLevel] = nFilesOnCard;
        } else if(menuPos[menuLevel] > 0) {
            if(menuPos[menuLevel] > 1 /* abs(next) */)
                menuPos[menuLevel] -= 1; // abs(next);
            else
                menuPos[menuLevel] = 0;
        }

```
and replace the 1 with abs(next) which you see in comments to get old behaviour.