repetier/RepetierFirmware #711: ATC Semitec 104GT-2 Temptable

Nibbels, 2017-10-30:
I have been dropping the temptable to Excel and saw some Spike.
[image]
222 | 1760

When I drop over Marlins values they dont have this spike:
[image]

Yes, I know how ridiculous this Issue is, but yeah ..
Might this cause some temperature shift around 210°C till 230°C ? 

Greetings

Nibbels, 2017-10-30:
Additional Info: 
We have a german thread on this. The source of the need to research this was, because autotune did not work as expected on sensor type 8
http://www.rf1000.de/viewtopic.php?f=23&t=2054&start=40#p20813

;)

repetier, 2017-10-30:
Thanks for the hint. Have corrected it for next dev update.

closed 2017-10-30 by repetier