repetier/RepetierFirmware #714: Feature Request: Safety Temperature Sensor

innovo3d, 2017-11-05:
Is there anyway we can add a safety temperature sensor?  

Reason: I just had a cool end fan fail on me.  My hot end target temperature of 390 degrees C was maintained and did not trigger anything suspicious, but since my cool end fan failed the rest of my extruder was also heating up... eventually it melted the plastic on the PC4-M6 bronze fitting and then went on to melt the plastic in the linear rail carriage that the extruder is mounted to...  I'm surprised that it didn't do more damage...

I have attached a provisional temperature sensor but realize that there's nothing in the firmware for this.  

Any chance we can get this in there?   Ideally, there is a safety max temp that if reached will cause an error and stop all heaters...

repetier, 2017-11-05:
I know a company using hardware temp break for this. If the temperature is reached, it will shut off the heater power as a second safety line.

Your idea is good for version 2 of firmware. There we will change the structure a bit making such an addition much easier. Have marked it as feature request.

innovo3d, 2017-11-19:
That's great.  Do you know when we can expect this version of the firmware
to be available?  I'll try to find another solution in the short term,
perhaps a temperature safety switch/relay as you have mentioned.

<https://mailtrack.io/> Sent with Mailtrack
<https://chrome.google.com/webstore/detail/mailtrack-for-gmail-inbox/ndnaehgpjlnokgebbaldlmgkapkpjkkb?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality>

*Adam Rumjahn*
*CTO & Founder*

* <http://www.innovo3d.co>*

*adam@innovo3d.co <adam@innovo3d.co>Cel: +57 304 643 1248*
*Fijo: +57 2 374 2473*

repetier, 2017-11-20:
No, at the moment we are quite busy on the server, so no time to start a new firmware version with new structure.