repetier/RepetierFirmware #717: compile error - #error STEP_DOUBLER_FREQUENCY should be in range 10000-16000.

sourianos, 2017-11-21:
Arduino IDE 1.8.5
[image] 

Deny compiling due to this error at repetier firmware files v0.92.

exit status 1
#error STEP_DOUBLER_FREQUENCY should be in range 10000-16000.

repetier, 2017-11-21:
It is no error, it is a sanity check. I bet your STEP_DOUBLER_FREQUENCY value is not in the range 10000-20000 so you try something that the cpu can not handle.

Nibbels, 2017-11-26:
Whenever there is a bit more code in ISR(TIMER1_COMPA_vect) even 10000 is not a good Idea. -> I have to use 8000-.
(Btw. Double Stepping and Quadstepping seem to work quite well since Halfstepping was removed.)

repetier, 2017-11-26:
You can simply remove the check if you like. In my tests I could get up to 15000Hz but that was caresian with no delays required. For other systems it might in deed need lower values.

sourianos, 2017-11-29:
Thank you very much!!!

repetier, 2017-11-29:
Reduced minimum in current dev.

closed 2017-11-29 by repetier