repetier/RepetierFirmware #718: Mks tft 32

zagranichnyy, 2017-12-12:
someone has encountered a problem, does not work in conjunction with the display mks tft32 all the configs configured correctly, with the computer prints without problems, but sending from the display to print does not work, in handall the configs are configured correctly, the computer prints without problems, but sending from the display to the print does not work, in manual mode all axes are moved, heating in

craftaz, 2017-12-28:
I have same problem too. No fix yet? Repetier 0.92.8 and MKS TFT32 v3

iosonopersia, 2017-12-28:
The problem is the outdated serial protocol that the firmware inside that display uses. It cannot communicate correctly with the printer, I think this has nothing to do with Repetier Firmware. Also a few MK4duo and Marlin users experienced this problem... I believe it's up to the MKS TFT display producers to fix this.

repetier, 2017-12-28:
Exactly. The display works like a host using serial communication. So it must support a firmware like hosts do. If they can not handle the format they simply will not work. Unfortuantely it is closed source so no one can see what is going on there.

closed 2017-12-28 by repetier