repetier/RepetierFirmware #720: Using Repetier for laser/cnc without extruder show  extruder menu

luc-github, 2017-12-19:
Hi, I am trying to setup a laser / cnc  with repetier and it show extruder menu - but I have none, so it is confusing

I have pushed a PR https://github.com/repetier/Repetier-Firmware/pull/719

Hope is it ok 

I think EEPROM export related options and Filament printed length should also be hidden in that case
I can do another PR for this also if it is Ok

repetier, 2017-12-19:
PR is merged.

You could do it, but please only reduce for the output not creation side. That way hosts do not show unnecessary options but can still set them if needed.

luc-github, 2017-12-19:
Yes - only to output - I do not want to mess with EEPROM mapping if user change his configuration 

thanks for the confirmation - I will work on it and push it for review later

luc-github, 2017-12-22:
PR done - https://github.com/repetier/Repetier-Firmware/pull/722
I think I did not missed anything

repetier, 2017-12-22:
Yes, looks complete. Closing issue. Thanks for improvement.

closed 2017-12-22 by repetier