repetier/RepetierFirmware #721: M408 JSON output repetier fw dev version

svs072477, 2017-12-22:
Comand M408 output not correct.
21:53:58.581 : {"status": "I","heaters":[22.638889,22.777778],"active":[0.000000,0.000000],"standby":[0.000000,0.000000],**"hstat":[00.000000,0],**"pos":[0.000000,0.000000,0.000000]**,"extr":[],"sfactor":100,"efactor":[100],"tool":0,"probe":1000,**"fanPercent":[0.000000],"homed":[0,0,0],"coords": {"axesHomed":[0,0,0],"extr":[0.000000],"xyz":[0.000000,0.000000,0.000000]},"currentTool":0,"params": {"atxPower":1,"fanPercent":[0.000000],"speedFactor":100,"extrFactors":[100]},"temps": {"bed": {"current":22.638889,"active":0.000000,"state":1},"heads": {"current": [22.777778],"active": [0.000000],"state": [1]}},"time":390291,"axisMins":[0,0,0],"axisMaxes":[170,170,200],"accelerations":[1000.000000,1000.000000,100.000000,5000.000000],"firmwareElectronics":"Arduino Due","firmwareName":"Repetier","firmwareVersion":"1.0.0dev","minFeedrates":[0,0,0,0],"maxFeedrates":[200.000000,200.000000,20.000000,50.000000]}

In ver 0.9.29 it's ok

repetier, 2017-12-22:
Since you know correct answer, what would it be resp. what is the error here?

svs072477, 2017-12-22:
"hstat":[00.000000,0]
"extr":[],
The devices heater0, extruder 0 are present, working. json shows that there are none

repetier, 2017-12-26:
Ok, found the problem and have fixed it.

closed 2017-12-26 by repetier