repetier/RepetierFirmware #723: Not able to connect after firmware update

jstockton, 2017-12-22:
I have a Tronxy/Zonestar P802M and recently updated the firmware to 0.92.9 and now cannot connect to my printer through either Repetier Host or Octopi.  I didn't see anything in the logs that would indicate any issues, but could have missed something or not looked in the correct spot for logs.  

Any suggestions on what I can do to fix the connection issue?

repetier, 2017-12-22:
New update might have set a different baud rate causing the connection problem. So fix is use new baud rate. If it has a lcd you can see baud rate in config.

jstockton, 2017-12-22:
Thanks for the idea.  I checked the baud rates before I flashed and they matched.  For kicks I tried changing it both on the printer and OctoPi to see if that got any results.  I dropped it to 9600 and still had the same results.

I do have a LCD so I can change some settings, but since I can't connect via USB (tried a new USB cable as well) I won't be able to flash any different settings.

repetier, 2017-12-22:
When you change in LCD you need to select store to eeprom afterwards or the setting will be lost on next reset which is required to make the change effective. Normally on connect it resets but some boards don't. So make sure you see in lcd a reset when connection is tried and check baud rate then to see if it was really taken.

jstockton, 2017-12-22:
I did store to EEPROM and confirmed the setting stuck by rebooting the printer.  Original baud was 115200.  I tried 9600 and 250000.  No luck.

I did notice some messed up information in the log that I didn't see before:

2017-12-22 22:45:52,195 - Connecting to: /dev/ttyUSB0
2017-12-22 22:45:52,206 - Changing monitoring state from 'Offline' to 'Opening serial port'
2017-12-22 22:45:52,211 - Connected to: Serial<id=0x70e405f0, open=True>(port='/dev/ttyUSB0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2017-12-22 22:45:52,211 - Changing monitoring state from 'Opening serial port' to 'Connecting'
2017-12-22 22:45:52,216 - Send: N0 M110 N0*125
2017-12-22 22:46:17,861 - Recv: ��DC������$��b������������������t�������t�������t�������t�������t�������t�������t�������t�����������$@���E#������D#�E�����D������b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��	��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��
2017-12-22 22:46:27,380 - Recv: ��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��
2017-12-22 22:50:01,194 - Recv: ��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b5g���b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��1��b��b��b��b��b��b��b��b��b��b��b��b��b�C1��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b�C1��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��1��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b������b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b�C1��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��
2017-12-22 22:50:54,274 - Recv: ��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b!�1��b��b��b��b��	��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��
2017-12-22 22:51:23,820 - Recv: ��b��b��1��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��b��

shaiss, 2019-06-12:
under serial connection > Firmware & Protocal > Protocal Fine Tuning > Blocked Commands add M110.  Alternatively, M110 could be added to repetier fw to have the same behaviour like marlin.