repetier/RepetierFirmware #725: ui.cpp addFloat with digits>3 -> possible uncaught overflow

Nibbels, 2018-01-03:
https://github.com/repetier/Repetier-Firmware/blob/95b5a979426176ed908910635e70635f6f900952/src/ArduinoAVR/Repetier/ui.cpp#L1171
vs.
const float roundingTable[] PROGMEM = {0.5, 0.05, 0.005, 0.0005};

It took me some while. I wanted to display some tiny number and stumbled upon weird behaviour.
Maybe you never need this hint, but some limiter for "digits" would be safer for future coding:

digits = (digits <= 3 ? digits : 3);

See
https://github.com/Nibbels/Repetier-Firmware/commit/8dfca1388a6a94291938f756e69bfe077de6b854

Greetings

repetier, 2018-01-03:
const float roundingTable[] PROGMEM = {0.5, 0.05, 0.005, 0.0005, 0.00005, , 0.000005};

is a better solution. tests take time and with correct programming no tests are required.

closed 2018-01-03 by repetier

Nibbels, 2018-01-04:
Yes,
 you would probably not make the mistake because you know the function.
I just wanted to tell you that there is some function which has a questionable integrity and wouldnt withstand against fuzzying.

Maybe some inline warning near the functions definition would help to prevent future missuses. (and not temper with execution speed)

Greetings