repetier/RepetierFirmware #728: Feature request

jamesarm97, 2018-01-08:
Is there some way to have a bed leveling assistant like Marlin has that will move to three defined points and then move to Z0 so you can manually adjust the bed leveling before doing the auto leveling or if you are not using auto leveling? It would need to pause and wait for user input before moving to each point and also lift the nozzle some before moving so it wont crash. I have something similar I have done using gcode but it has to be run from the SD due to the pause command and the screen always changes once paused so you can't see the instruction message.

boelle, 2018-01-08:
you could have a gocde script on the sd card that does the same

i have a script for changing filament since it was more easy for me and my "long" bowden tube

it would be easy to make a script that moves to those points and go to Z0 and after the last point it could start auto leveling

having a script on the sd makes the firmware less cluttered

jamesarm97, 2018-01-09:
That is what I have already (on SD). It is just something the old firmware had built in but I hate Marlin and it was just not working out so I switched to repetier and will be getting the firmware out for others to use.

boelle, 2018-01-09:
If you have it on SD allready then why clutter the firmware with the same thing?

repetier, 2018-01-09:
I don't have a marlin delta printer, so I don't know the wizard.

What does it do with the 3 points it moves to? What problem does it solve?

Currently we have no such wizard. New features will only be added in V2.0 once the conversion is finished.

luc-github, 2018-01-09:
I have implemented the manual bed leveling wizard for the  Davinci fork : https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/ui.cpp#L5682-L6031 if it can help

jamesarm97, 2018-01-09:
That is exactly what I am looking for. Thanks. Just an easy way to "level" the bed they way it should be before enabling the Auto Leveling with probe. I know you can just home and move the head around manually but I was hoping to keep the same functionality when I offer new firmware to replace the junk that came with the printer.

luc-github, 2018-01-09:
@jamesarm97  keep in mind you cannot merge 'it as it is' because it depend of several ui functions and wizard functions not present in original repetier,  you will have to adapt the code if you want to add it

jamesarm97, 2018-01-09:
Understood. I just forked the Repetier repo so I can start merging in your bed leveling changes for me to use. I should be able to find the parts needed just for this.

boelle, 2018-01-09:
if you get it working maybe @repetier would take the changes as a PR for the dev branch?

boelle, 2018-01-15:
maybe this is a good one for V2 of the firmware? ie put a V2 tag on it if the PR comes in

repetier, 2018-01-15:
Ok, marked it as feature request.