repetier/RepetierFirmware #730: G29 grid measurements like G32

cinhcet, 2018-01-11:
Hi,

I really like the repetier stuff, from the firmware over the host to the server!

Z probing on grid points is only supported for G32 and not for G29. Is there a particular reason? 
I'm asking because I would like to level my bed manually by using the z-probing feature for the measurements with G29. When using G32 for that, one would have to disable the autoleveling afterwards.

Version 1.0.0.

(Another small thing I find strange is that the probe position points are the positions of the actual probe. Wouldn't it make more sense to use the origin of the extruders, because this is what can be seen when controlling the printer and the probe offset has to be defined anyhow.)

jamesarm97, 2018-01-11:
I'm pretty sure you can configure the X / Y / Z offset of the probe from the Extruder and when you tell it to probe a position it will add those offsets and go to the correct place. So when you move the extruder manually to see what position it is, it will show the extruder position but add the correct offset before moving to that point during the probing (At least that is how I understood it).

cinhcet, 2018-01-11:
yes, this is what I think is strange :-) Since there is an offset, I do not want the probe to go where the extruder is at the moment. Instead, it should just go where I wanted it to be at the moment of the position read out. But I understand that the ability to set the real z probe position is also reasonable. Thanks

repetier, 2018-01-11:
G29 was always meant as more accurate homing replacement to find an average height. Not sure why this was wanted. So it just measures an average height and sets it. 

G32 is for autoleveling and you can disable it afterwards if wanted (M321) so that is no problem.

None of them is meant for manual bed leveling. Repetier-Host can measure a grid without any changes if you want to use that, but for manual leveling you need exactly 3 points to define the rotation. Everything else is just to remove measurement errors and might confuse more then help.

All tools have an offset to a virtual origin if your slider, normally extruder 0 is the reference point and xy probe offset are relative distances to that. On activating the probe it will move such that the z probe is at then given position and not the previously active extruder.

cinhcet, 2018-01-11:
thanks for your fast answer. 

I was not aware of that feature in repetier-host. Very nice! I took my info from https://www.repetier.com/documentation/repetier-firmware/z-probing/ where it says at the very end of the page to use G29 for manual bed leveling.

I still think that if G29 behaves like G32 would be more consistent. Initially, I thought that my configuration must be wrong. Three points for beds that are mounted with four screws are not ideal, in my experience. 
But anyway, its not a big deal....

repetier, 2018-01-11:
With 4 screws it is in deed more stable but you always try to bend the bed. It becomes apparent if you put one extremely or loosen one completely, it is still more or less table. So you need to adjust on 3 screws and then tighten the 4th just to be at the given height.

Maybe I update G29 in V2 to test the same points as G32, but V1 is finished.

closed 2018-01-11 by repetier