repetier/RepetierFirmware #731: Filament Change, double retraction

PvtPrivate, 2018-01-11:
When changing filament, after priming the nozzle with the encoder to turn the extruder and then clicking the button it's supposed to do a retract, move to the last position on the print, then unretract, then resume printing.

But when retracting, it does a "double retract". A normal retract (4mm), then when the head is moving to the print, it also moves the extruder back (for an unknown amount). When the head is back on the print, it unretracts (4mm) then resumes printing. The second retract never got unretracted, thus leaving a layer on the print without filament.

This also applies when there is a jam.

https://www.youtube.com/watch?v=TGwXV6a2dy4

jamesarm97, 2018-01-11:
If it is related to anything I have seen, then it is because you manually extruded and then resumed without setting the E value back to 0 in the menu. That is what I found anyways not sure if it is related.

PvtPrivate, 2018-01-11:
I extruded using the encoder in the filament change wizard.

repetier, 2018-01-12:
I think I see the error. In Printer.cpp moveToReal it should look like this
```
uint8_t Printer::moveToReal(float x, float y, float z, float e, float f, bool pathOptimize) {
    if(x == IGNORE_COORDINATE)
        x = currentPosition[X_AXIS];
    else
        currentPosition[X_AXIS] = x;
    if(y == IGNORE_COORDINATE)
        y = currentPosition[Y_AXIS];
    else
        currentPosition[Y_AXIS] = y;
    if(z == IGNORE_COORDINATE)
        z = currentPosition[Z_AXIS];
    else
        currentPosition[Z_AXIS] = z;
    transformToPrinter(x + Printer::offsetX, y + Printer::offsetY, z + Printer::offsetZ, x, y, z);
    z += offsetZ2;
    // There was conflicting use of IGNOR_COORDINATE
    destinationSteps[X_AXIS] = static_cast<int32_t>(floor(x * axisStepsPerMM[X_AXIS] + 0.5f));
    destinationSteps[Y_AXIS] = static_cast<int32_t>(floor(y * axisStepsPerMM[Y_AXIS] + 0.5f));
    destinationSteps[Z_AXIS] = static_cast<int32_t>(floor(z * axisStepsPerMM[Z_AXIS] + 0.5f));
    if(e != IGNORE_COORDINATE && !Printer::debugDryrun()
#if MIN_EXTRUDER_TEMP > 30
            && (Extruder::current->tempControl.currentTemperatureC > MIN_EXTRUDER_TEMP || Printer::isColdExtrusionAllowed() || Extruder::current->tempControl.sensorType == 0)
#endif
      ) {
        destinationSteps[E_AXIS] = e * axisStepsPerMM[E_AXIS];
    } else {
		destinationSteps[E_AXIS] = currentPositionSteps[E_AXIS];
	}
```

I added here the last else. The previous retract was relative in steps so it set destination to be different from current and the xy move will not change e so destination for E is not set and the old difference is used again.

Will publish a patch at the weekend along with some other changes. But you can modify the function already to verify. Never noticed this on my printer, but I also have not such a nice view on extruder movements like you have.

PvtPrivate, 2018-01-14:
I can confirm the above fix works.