repetier/RepetierFirmware #739: Feature request V2: Add support for Mini Height Sensor Board

boelle, 2018-01-15:
https://miscsolutions.wordpress.com/mini-height-sensor-board/

important notes from that link:

- With the sensor a long way above any surface, the reading should be close to zero.

- With the sensor close to a surface but slightly further away than the trigger height, the reading should be about 465.

- With the sensor slightly closer to a surface, the reading should be about 535 and the red LED on the sensor board should illuminate.

- If you place a surface right up against the sensor board, then the reading may drop to near zero again. This is normal.

- If you see a reading of around 1000, or if the LED flashes rapidly, this means that there is too much ambient IR for the sensor to function correctly. This normally happens only when direct bright sunlight is reflected from the bed into the sensor, or when you place a highly reflective surface below the sensor such as aluminium foil.

repetier, 2018-01-15:
It also has a digital output mode so it can already be used if you want.

boelle, 2018-01-15:
that is true, but i guess the analog mode could be used to get a more precise result. or am i totally mistaken?

it will take a bit longer but might be worth it

repetier, 2018-01-16:
The only advantage of analog mode is that you could start with higher speed and then slow down. But since it is analog you have to slow down because analog measurements are not as fast as digital signals. So the one advantage is only virtual and I would assume result is the same.

closed 2019-03-19 by boelle