repetier/RepetierFirmware #740: Feature request V2: Add support for Filament width sensor

boelle, 2018-01-15:
http://objectswithintelligence.weebly.com/store.html

it ouputs an analog voltage direct prob with filament width

a pot meter could be used for tetsting or a 1.5V battery

it would also need to have a setting in config tool that tells how far from nozzle the sensor is located

repetier, 2018-01-15:
Question is here how fast does diameter change and how reliable is the analog value. I mean if you measure 1.78mm thickness you might need to extruder 50-100mm until it has an effect an dat that time measures 1.76mm. So how many values do I need to store additionally to make the measured value have an effect on the result. And how much will that slow down everything.

For a constant value or nearly constant it is easy to just modify a multiplier for flow just like flow multiplier. But the delay is what might be needed to get any effect and of course bad filament to need it:-)

Do you know any test results with before and after and if it gives a visible difference? Especially if the result depends on things like how translucent filament is it do more worse then bad in the end.

boelle, 2018-01-15:
if you mean constant distance between probe and nozzle then yes, i would not imagine the setup done in any other way

there are a test video somewhere but will have to search for it

boelle, 2018-01-15:
https://www.youtube.com/watch?v=5JmroyGb4qY

demo print starts at 11:06

closed 2019-03-19 by boelle