repetier/RepetierFirmware #742: MKS SBase V1.3 with LCD32 issues with Repetier

aceranic, 2018-01-17:
I have issue with extruder temp indication in Repetier.
Temp reading is fine on MKS TFT 32 but it is frozen in Repetier. It takes reading when connecting and after that nothing. Stays the same.
Printer is connecting fine and working fine from Repetier.
Anybody have/had same.

repetier, 2018-01-17:
Where is the temperature reading frozen? Do you have 2 display - mks plus lcd on firmware or is repetier one of our 3 host products?

aceranic, 2018-01-17:
I have MKS TFT display and reading is good there.
On PC Repetier reading is frozen

repetier, 2018-01-17:
On PC means repetier-host I assume? Is only reading frozen or does connection not work at all. I mean can you still move it around? Would firmware respond to M119 and report endstop status in log?

aceranic, 2018-01-17:
Everything is working except temp reading.
M119 is working too

repetier, 2018-01-17:
Check printer settings. There is a checkbox to enable M105 temperature query. Guess it is simply off so host will not ask for temperatures and you see nothing.

aceranic, 2018-01-17:
Yes that was it. Thanks a lot

closed 2018-01-17 by repetier