repetier/RepetierFirmware #743: Is there a "plug and play" configuration for download for RAMPS 1.4 please

Fireflaker, 2018-01-17:
Please......... I tried and there is no display on LCD.

Fireflaker, 2018-01-17:
Also, for a Delta printer.

repetier, 2018-01-18:
There are millions of possible configurations and without answering 200 questions it is impossible to find the right config.

Extruder normally work only when hot or cold extrusion is allowed (M302 S1). Also see in eeprom if steps per mm etc have good values.

boelle, 2018-02-16:
@Fireflaker 

here is my configuration.h you can use that as an example and it comes without any support (i dont have the time)

https://pastebin.com/2pLSFj33

save the content of the pastebin as configuration.h and then go to https://www.repetier.com/firmware/dev/index.php and load it, then change settings

i have this little fan extender also: https://www.reprap.me/rrd-fan-extender.html

i use this to control some extra fans, one to blow on the electronics if steppers or FET's are active, one to blow on bed after print is done to get it of the pei quicker. and the fan for the E3D comes on when temp is above 50C

badvision, 2018-03-08:
I have Ramps 1.4 configured just fine in my printer -- I didn't find it challenging to set up.  However, every printer has differences such as dimensions and steps per mm settings, etc.   Unless you took a MakerSelect v2.1 printer and replaced the Melzi with Ramps, I doubt my settings file will be very useful to you.